JetBrains/intellij-micropython #1: No autocompletion for display.<func>

lancelote, 2017-06-24:
PyCharm suggests no autocompletion for `display` objects, e.g.:

```python
display.scr  # no suggestion "scroll"
```

also `scroll` and other `display` functions have no parameter hints. The problem seems to be with alias imports in [__init__](https://github.com/vlasovskikh/intellij-micropython/blob/master/typehints/microbit/latest/microbit/__init__.pyi#L31) file.

Changing from:

```python
from . import (display as display, ...
```

to

```python
from . import display, ...
```

fixes the problem. So I wonder - why are the alias imports used there in a first place?

vlasovskikh, 2017-06-25:
According to [PEP 484](https://www.python.org/dev/peps/pep-0484/#stub-files) Python stubs require re-importing:

> Modules and variables imported into the stub are not considered exported from the stub unless the import uses the import ... as ... form or the equivalent from ... import ... as ... form.

I believe it's a bug in the code insight engine of PyCharm not related to MicroPython. Could you please report it to its issue tracker?

lancelote, 2017-06-25:
I will, thank you for an explanation! Please feel free to close.

closed 2017-06-25 by vlasovskikh

lancelote, 2017-06-26:
Ticket https://youtrack.jetbrains.com/issue/PY-24891

*Note:* I failed to reproduce the problem outside of MicroPython support.