JetBrains/intellij-micropython #2: Framework to support ESP8266 devices

vlasovskikh, 2017-06-26:
There are many devices based on the ESP8266 chip:

*  Adafruit Feather Huzzah ([link](https://www.adafruit.com/product/2821))
* Adafruit Huzzah ([link](https://www.adafruit.com/product/2471))
* Wemos D1 Mini ([link](https://wiki.wemos.cc/products:d1:d1_mini))
* Wemos D1 Mini Pro ([link](https://wiki.wemos.cc/products:d1:d1_mini_pro))
* NodeMCU

and others.

It would be nice to have them supported by the MicroPython Support plugin for PyCharm / IntelliJ.

See the discussion about supporting ESP8266 in PyCharm at [PY-20308](https://youtrack.jetbrains.com/issue/PY-20308).

This issue is about the initial support for the ESP8266 devices.

vlasovskikh, 2017-06-26:
I've got Wemos D1 mini. I'll start working on the experimental support for it.

The current framework lacks the ability to permanently save the device selected in the configuration dialog. It's also not possible to add per-device UI elements to the configuration dialog. I've added these items to the roadmap in the issue text.

I found it hard to setup my device since the official website lacked the link to the up-to-date driver for the serial port of Wemos D1 mini. I've stated a [wiki page](https://github.com/vlasovskikh/intellij-micropython/wiki/Wemos-D1-mini) about the device.

costastf, 2017-06-26:
Ηι @vlasovskikh . Keep in mind that for all the boards out there, there are also a ton of clones that might behave slightly differently. I have a few wemos d1 mini and while some flash just fine with:
```esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=detect 0 MICROPYTHON_FIRMWARE_FOR_ESP8266.bin```
some of them just stay with their led on with the preceding command and require 
```esptool.py --port /dev/ttyUSB0 write_flash -fm dio -fs 32m 0 MICROPYTHON_FIRMWARE_FOR_ESP8266.bin```
to properly flash.

lensvol, 2017-06-26:
@costastf I think the situation you are mentioning here is a well-known issue with WeMOS d1 mini _Pro_, related to increased flash size (16Mb) -
 https://github.com/micropython/micropython/issues/2335

It is a separate WeMOS board and should not be considered a clone. Otherwise, you are right on point. Perhaps there are should be an option to manually edit arguments to esptool? Assuming there will be esptool support at all.

vlasovskikh, 2017-06-26:
@costastf Right, there could be various configuration parameters. I would like to start with almost no user-visible options in the plugin besides the precise model of the device. In theory, the model should be enough to be able to provide the necessary parameters or auto-detect the missing parameters like the device port. If the precise device model turns out to be not enough, I will reconsider this approach.

costastf, 2017-06-26:
Hi, @lensvol in my case all boards have the same amount of memory and they are not the "pro" board. The same thing happens with some esp01 that I have, where the "on" led is inverted and on some of them it stays on and on some of them only comes on when writing. There is a big market in China with clones and they are very very difficult to identify and sometimes they do respond differently. For extra details you can have a look at http://hackaday.com/2017/05/15/attack-on-the-clones-a-review-of-two-common-esp8266-mini-d1-boards/

lensvol, 2017-06-26:
Hi @costastf, I look into that further, thanks. It just reminded me of a similar issue with Pro version, that looked exactly like that and was fixed using command specified above (it was with stable 1.9.1, I believe).

vlasovskikh, 2017-06-26:
@costastf @lensvol USB VID:PID numbers for the USB controllers of MicroPython boards could be enough for distinguishing the connected boards among all the USB devices. I've implemented WEMOS D1 mini REPL using this technique. The current version of the script is able to detect both BBC Micro:bit and WEMOS D1 mini based on their VID:PID. See #3 and our [WEMOS D1 mini|https://github.com/vlasovskikh/intellij-micropython/wiki/WEMOS-D1-mini] wiki page.

eduvik, 2017-06-26:
@vlasovskikh the USB IDs are probably not unique to micropython boards as they're just USB-serial chips. Additionally, there's the use case of a 'bare' ESP board without onboard USB at all (e.g. ESP01, ESP12 etc) that need an external serial adapter. Having said that, simply the presence of a serial adapter on the machine will be a pretty big clue there's a device connected for _most_ (not all) people. Detecting the devices by polling the serial ports might be possible, but not 100%. I'd be inclined to implement base functionality first (user chooses the model and the serial port it's connected to) before trying the autodetection.

vlasovskikh, 2017-06-26:
@eduvik The port name might change depending on which USB port you used for connecting your device. Auto-detection could be handy in this case.

Since auto-detection already works at least for WEMOS D1 mini, I'm going to proceed with other features for this device, see the roadmap at #3. Later on I might return to adding the field for the port name combining it with auto-detection capabilities.

costastf, 2017-06-27:
Hi, @vlasovskikh I think that @eduvik is right about the autodetection. A wemos I happen to have with me shows up on my linux box as a ch341-uart serial to usb converter which of course is a very very generic component and I think that that would probably cause issues in the long run. Just for reference it shows up under lsusb as ```Bus 001 Device 009: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter```

Also as mentioned esp01 do not have a usb interface at all and they are powered with 3.3 so they definitely require an adapter which means there is no way to auto detect them. I think the point here is that people who fiddle with stuff like that should have at least the basic understanding of serial, usb and such so autodetection might produce a lot of false positives and issues without providing too much in return. As @eduvik said I am all for trying to go for basic functionality first. 
Any news on the webcast for the development environment? I work with wemos d1 so I would be eager to help in any way I can.

vlasovskikh, 2017-06-27:
Ok, let's add a field for entering the device port manually. Still I prefer it to be auto-detected if it's empty.

vlasovskikh, 2017-06-27:
I need your help with a couple of tasks:

* Type hints for MicroPython and ESP8266 modules
* A tool for flashing files to WEMOS boards (maybe a modification of what `ampy` does). As I've mentioned in #3, `ampy` doesn't work with them

Please put a comment here if you'd like to contribute any parts of these.

vlasovskikh, 2017-06-27:
@costastf I caught a cold, so won't be able to do the webcast in the next couple of days.

vlasovskikh, 2017-06-27:
@costastf Have you managed to launch IntelliJ 2017.1 and import the project into it? I could answer your questions about the development environment setup here in the comments.

costastf, 2017-06-28:
Hi Andrey,

Something personal came up and I will not be able to help out until next
Tuesday. I am very invested on this and coming next week I will come on
board and help out any way I can.

Kind regards
Costas

vlasovskikh, 2017-06-29:
We've just got an experimental provider for WEMOS D1 mini Pro. Thanks @lensvol for your PR #4!

vlasovskikh, 2017-06-29:
I've created separate issues for:

* Type hints for MicroPython modules #5 
* Flash Python files to ESP8266 devices #6

eduvik, 2017-06-29:
Nice to see movement happening, but I still think it's the wrong approach to use USB IDs here, for the following reasons:

- there are more ESP8266 devices than we'll be able to discover and test, but in theory all of them should work pretty much identically (with the possible exception of flashing micropython in the first place). Maintaining that list will not be fun.
- ESP devices without an onboard USB-serial will not be picked up by this. Conversely, the USB ID of many USB-serial chips on ESP boards will be the same as external USB-serial adapters, so simply the presence of a USB-serial adapter plugged in to the computer will create a false positive. 
- What happens when two identical ESP8266 devices (e.g. WeMOS d1 mini's) are connected to the computer? 

I would suggest the following approach:

- enumerate the serial ports available on the computer. 
- If there's only one, great, that's probably it. Now we can attempt to identify the type of board. USB ID's might be part of that mix, but as said, is not reliable, and prone to maintenance problems. It might be better to initiate communication with the port, check if it's micropython, version, etc (can't remember what info is presented).
- If there's more than one serial port, I'd probably suggest deferring to settings. It would be too easy to write to the wrong board if the detection gets it wrong. Another thought is to attempt to get the device to flash an on-board LED, but that's assuming every device has a LED (I don't think they all do, and they're configured in different ways). Let's leave the LED bit as a stretch goal.

vlasovskikh, 2017-06-29:
@eduvik Let's discuss it in a separate issue #7.

vlasovskikh, 2017-07-19:
FYI Last weekend I gave a talk on MicroPython mentioning this plugin. Here are the slides: https://twitter.com/vlasovskikh/status/886600153767038977

elf55, 2017-07-20:
Well done.  interesting details about uPython memory usage vs Python.   

-Mike

vlasovskikh, 2017-09-04:
Meanwhile we've got basic support for Pyboard thanks to @stefanhoelzl.

You can try the plugin with ESP 8266 and Pyboard support from the master branch. What features are missing?

vlasovskikh, 2018-01-15:
MicroPython plugin 1.0 provides initial support for ESP8266-based devices. Please send your feedback in new issues / pull requests.

closed 2018-01-15 by vlasovskikh