JetBrains/intellij-micropython #3: Support Wemos D1 mini

vlasovskikh, 2017-06-26:
See also #2 about the common framework for ESP8266 devices.

The wiki page for this device is [here](https://github.com/vlasovskikh/intellij-micropython/wiki/Wemos-D1-mini).

vlasovskikh, 2017-06-26:
I'm unable to connect to my Wemos D1 mini using `ampy` from `adafruit-ampy` even though the REPL protocol is the same. Some further investigation is required.

vlasovskikh, 2017-06-26:
I've added REPL over USB for WEMOS D1 mini.

vlasovskikh, 2017-06-27:
I was able to exec some code on my device using `pyboard.py` (a tool originated in the MicroPython codebase and included into `ampy` as the basis for executing commands. The only thing is that I had to wait at least 0.5 sec between establishing a serial connection and entering a raw REPL mode:

```python
from ampy.pyboard import Pyboard
from time import sleep

program = """\
from machine import Pin
from time import sleep

led = Pin(2, Pin.OUT)

while True:
    led.off()
    sleep(1)
    led.on()
    sleep(1)
"""

b = Pyboard('/dev/cu.wchusbserialfd120')
sleep(0.5)  # Had to wait for >= 0.2 sec in order to be able to use raw REPL
b.enter_raw_repl()
b.exec(program)
b.exit_raw_repl()
```

I believe that the output expected by `pyboard.py` depends on timing, while it shouldn't in general. I'm thinking of either working around this problem by actually adding this `sleep()` into my patched version or spending some time determining the cause in the current communications protocol.

costastf, 2017-07-10:
Hi @vlasovskikh . I have just managed to set up a machine for me and I am trying to figure out how to setup a development environment to give a hand with this. If you would still be willing to have a webcast on that it would be great, I have no time constrains, any time that would be good for you i can make it. I followed the small documentation that you have posted and I run the gradle and it build an intellij instance. Not having worked with java or any of its derivatives before I am a bit lost i am afraid. Any help would be greatly appreciated.

vlasovskikh, 2017-07-12:
@costastf I'll be quite busy till Monday, sorry about it. Let's get back to it on Tuesday.

vlasovskikh, 2017-08-17:
Added the ability to flash files and folders to WEMOS D1 mini. I will fix some bugs and generalise it to other ESP devices.

vlasovskikh, 2017-09-05:
I've merged WEMOS D1 mini and WEMOS D1 mini Pro into a single device provider for ESP8266 devices as various devices appear to have no significant differences from the IDE point of view. See #2.

closed 2017-09-05 by vlasovskikh