JetBrains/intellij-micropython #6: Flash Python files to ESP8266

vlasovskikh, 2017-06-29:
1. There are several possible modes of operation:

* Upload a single file path/to/file.py as main.py
* Upload a single file to its corresponding path path/to/file.py
* Upload every file under directory path/to/directory to the device

Which one would you prefer?

2. ESP8266 makes it possible to upload files via:

* Serial USB REPL
* WebREPL

Any preferences here?

3. Are you OK with using a Run Configuration in PyCharm / IntelliJ for flashing Python files to devices? The program is only flashed to the device and runs after its reset. The Run Console in the IDE would show only the results of the uploading process. Should it be a separate upload action instead?

* A Run Configuration is OK
* A separate action in the menu would better reflect the semantics of uploading Python files

vlasovskikh, 2017-06-29:
Uploading could be done via the `adafruit-ampy` package. Although I've experienced some issues using it with WEMOS D1 mini, see #3. At least MicroPython's `pyboard.py` shipped as a submodule of `ampy` could be helpful as the core library for uploading files via REPL.

elf55, 2017-06-29:
I would prefer:

1. upload multiple files from a folder.
2.  upload via WebREPL.  But either is OK.
3.  Separate upload and run steps.  It's sometimes useful to upload and make an adjustment via REPL before running.

lensvol, 2017-06-29:
1. Single file and/or multiple files. 
2. Upload via USB cable. WebREPL needs to be explicitly enabled, can be fickle and needs a presence on the same Wi-Fi network.
3. Separate upload action seems like a good choice. "Run" step mainly amounts to performing a soft reset on a board, anyway.

vlasovskikh, 2017-07-01:
Check out my draft script for uploading a directory to a MicroPython device 8be5cea8c08a9db6af839748a69c97f25952df22. Try it from the command line (requirements: `adafruit-ampy`, `docopt`). Does it work for you?

lensvol, 2017-07-02:
It works with my mini Pro well enough.

Some thoughts:
* It needs to support uploading individual files along with whole directories. It could be convenient to flash just a single file with an update to existing code, instead of waiting for a full directory.
* How do you plan to handle uploading files while working in "Micropython REPL"? At this moment it is a hassle because ampy competes with the terminal application for the access to the same port.
* Hardcoding a single wait operation can backfire, in case some boards require just a little bit more time to become ready for REPL (e.g. 0.7 seconds). At the same time, too long of a waiting period can make things unwieldy. Maybe we can busy loop for small slices of time (0.1) while checking board status and counting towards whole timeout?

vlasovskikh, 2017-07-04:
@lensvol I will add the ability to upload a single file. I don't have any solution in mind for uploading files while the REPL session is active. If you have any ideas, please share them :) Let's start with the hard-coded timeout and make it configurable in some way later on, as it doesn't look like the highest priority issue.

lensvol, 2017-07-06:
@vlasovskikh The only solution I can think of right now:

1. Temporarily close REPL process (without closing Terminal View)
2. Run upload command with output redirected into Terminal View.
3. Restart REPL and reconnect it to the previously opened terminal window.

I am not that familiar with plugin development and IDEA internals, so I don't know if it's real or not. :)

vlasovskikh, 2017-08-17:
Ok, currently I'm doing it via the Run configuration "Flash ... to device". It supports flashing files and folders to WEMOS D1 mini. I'm going to generalise it. Feel free to open new tickets to discuss flashing options / implementation details.

closed 2017-08-17 by vlasovskikh