JetBrains/intellij-micropython #7: Option to specify the device port manually

vlasovskikh, 2017-06-29:
Several people in #2 have pointed out that auto-detection based on USB IDs of the serial controllers of devices may fail.

Even though I still think that auto-detection will work in most situations, we have to have a way to specify the port manually.

Open questions:

* The device settings are currently stored in the module settings which are likely to be put under VCS. The device port on the other hand is actually a very transient thing. Where should we store it? (this limits the places in the UI where it can appear)
* Where in the UI should this option appear? It is unlikely to appear in the MicroPython support settings due to the ever changing name of the port. If we put it in the Run Configuration options, we will be unable to use it in the REPL tool window which is invoked as a separate action.

vlasovskikh, 2017-06-29:
One of the options is to show an analog of the "Select Deployment Target" dialog with the UI and persistence options similar to [this Android Studio dialog](https://developer.android.com/studio/run/index.html).

This problem is also related to the way we will flash / run programs on devices, see #6 for the discussion.

lensvol, 2017-06-30:
Dedicated dialog with an option to save your choice for future use seems to be a good idea.

It will also allow us to abstract away from the notion of "connected via serial port" devices and support more exotic scenarios, like WebREPL or upload over FTP.

vlasovskikh, 2017-08-17:
Added the device path to the MicroPython settings.

closed 2017-08-17 by vlasovskikh