JetBrains/intellij-micropython #12: REPL doesn't start on Windows

eduvik, 2017-08-07:
When I try to start the micropython REPL on Windows (10 in this case, not tested on others), I get a traceback:   
```
File "C:\Users\xxx\AppData\Local\Programs\Python\Python36-32\lib\codecs.py", line 408, in __getattr__
    return getattr(self.stream, name)
AttributeError: 'Out' object has no attribute 'buffer'
```

When I try to run microrepl.py directly I can get the tracebacks; when I run it via the 'MicroPython REPL' menu in PyCharm it flashes an error then disappears.

The closest thing to this I could find online was https://github.com/wendlers/mpfshell/issues/5 which as to do with re-creation of the miniterm. I experimented with removing the two lines referring to Console, and I don't get that error any more (though the REPL doesn't work as expected).

vlasovskikh, 2017-08-17:
We currently don't support Windows. Feel free to send your PRs for Windows support.

eduvik, 2017-08-17:
Perhaps add that to the requirements list. Can you point me to which areas are OS-specific?

vlasovskikh, 2017-08-17:
Communicating with devices over USB using `pyserial` could be OS-dependent, I don't know if USB ports are represented via device files like /dev/tty.something.or.other. I can test this plugin only on macOS and Linux, so I won't be supporting it on Windows. But as I said, contributions are welcome.

jsbyysheng, 2017-09-14:
On windows, if u use `pyserial` to communicate, I think u can add codes like this to add support for windows.(serial device in windows is named like 'COM5')
```
# example for pyboard v1.1
import serial
import serial.tools.list_ports as stlp

# acquire all the serial devices
serialDevLst = stlp.comports()
# look for Python Board USB COM Port
for serialDev in serialDevLst:
    if 'Python Board' in serialDev.description:
        print(serialDev.description)
        pyb = serial.Serial(port = serialDev.device, baudrate = 4000000, timeout = 5)

pyb.close()  
```

vlasovskikh, 2017-10-20:
@jsbyysheng Sorry for the late reply. Since I have no easy way to test it on Windows, I would appreciate any pull requests that fix any Windows issues, as well as beta testers who are willing to try out these fixes.

handzsujt, 2018-01-17:
I moved one step further but I am not able to follow that for the moment:
According to the bug mentioned above (wendlers/mpfshell#5) it is necessary to change remove line 15 ("console = Console()") and change line 82 into "miniterm.console.setup()" because instantiation of "Miniterm" creates a "Console" object as an attribute of the "Miniterm" object. With this I can start the REPL but it shows only the starting output and is not reacting on any interaction. Sorry for not finishing this.

vlasovskikh, 2018-01-17:
@handzsujt Thanks for the idea! It was indeed the case. Actually, console initialization was performed three times, not just two. Also it's not possible to run this script in a regular PyCharm run configuration because it requires to be run inside a more or less real terminal. See 85de90c for my fix.

vlasovskikh, 2018-01-17:
Fixed in 1.0.4.

closed 2018-01-17 by vlasovskikh

kaybee335, 2018-01-18:
Just installed the plugin update I think I still have issues.  If I select the Python Console tab I see the REPL from what looks like my conda environment (similar to venv's) not the REPL from the board.  The Terminal tab has the usual CMD,EXE/PowerShell prompt (shudder - now that's where I would really like to see an ssh session to WSL!!).  The Run tab has the usual upload to board progress stuff in it.

Have I done something wrong or is it possibly the fact that I use conda rather than venvs that may be the issue?

vlasovskikh, 2018-01-18:
@kaybee335 Are you sure you have only one terminal tab in the terminal tool window? There should be two tabs: with your Windows shell and with a MicroPython REPL session. At the moment the plugin cannot control which one will be shown when this tool window. I hope I'll fix this tab focus issue later on.

MrYsLab, 2018-01-18:
@kaybee335  Did you select the REPL from the Tools/MicroPython/MicroPythhon REPL menu selection?
I just tried it on Windows 10, and I see both tabs.

kaybee335, 2018-01-20:
Yep - that's the one that shows me the command line from CMD.EXE almost like something failed to start and it fell back to CMD.EXE.  Is it possible that it is doing whatever it is doing, but NOT within the context of my conda environment?  If so, it won't find things such as adafruit-ampy and esptool which I supsect you use 'under the hood'.

The tab labeled Python Console shows me the command line from WITHIN the conda environment so I know it is active in PyCharm.

My best guess is that somehow things are being executed in my 'bare' CMD.EXE python environment and not the CMD.EXE as re-defined by activating the conda environment.  It fails because pre-requisites are not met in my 'bare' environment so it is falling back to CMD.EXE. There are no errors displayed in the CMD.EXE tab, it looks like a fresh start-up of CMD.EXE so it is also possible that it is an uninitialised CMD.EXE subprocess prompt being displayed on error conditions. 

I may be just blowing smoke here, it could be something  else entirely including problems between keyboard and chair....

kaybee335, 2018-01-20:
Could you please provide a more detailed explanation of the 'Remove All Files from Micropython Device' tool and exactly what it erases?  When I think 'erase flash', I think esptool so I am wondering if this will erase micropython itself (seems a bit drastic for an IDE :-) )

If it just erases 'files' om the device, does it erase /boot.py?  That would be slightly annoying but could be solved by just adding boot.py to each project after pulling it down from the board (probably a good idea anyway).

Does it remove subdirectories?  If so, is there any way to preserve them (frozen code, configuration files, etc.)?

Might be worth adding the info to the README as well just so people know.
Thanks.

kaybee335, 2018-01-20:
I'll answer my own question.  It erases all files, contents of subdirectories (I only tested one layer deep) and the subdirectories themselves.

It ALSO removes boot.py so you are going to want to grab a copy and add it to your project so that you can restore t later.

vlasovskikh, 2018-01-20:
@kaybee335 The issue with REPL shouldn't be related to Conda or your shell, since PyCharm launches just a Python process.

I'm afraid to debug this issue you have to try to launch the same command outside PyCharm as the plugin does. Use your Conda environment (where you've installed the package dependencies when PyCharm showed you a yellow warning about missing pyserial, adafriut-ampy, and docopt.

The command is: /path/to/your/virtualenv/or/conda/python.exe /path/to/your/pycharm/settings/.../microrepl.py your-device-name

Based on #35 I assume that the paths for you are:

C:\Users\...\condaenvs\upython\pythonw.exe C:\Users\...\.PyCharm2017.3\config\plugins\intellij-micropython/scripts/microrepl.py COM4

Please create new bugs for any issues not related to the problem with starting REPL on Windows like you did for boot.py in #35.