JetBrains/intellij-micropython #13: Failed to load plugin descriptor from file error

popeyesfx, 2017-08-17:
Tried to install plugin on 2017.2.1 version of pycharm, but get the above error.
Tried re-downloading it but still will not load

vlasovskikh, 2017-08-17:
How did you install the plugin? Is is the master branch from GitHub or the published version from the plugin repository?

popeyesfx, 2017-08-18:
i tried to install from jetbrains but it said that it was not compatible with latest version of pycharm, so downloaded zip from github and tried to install from that but got the error message

vlasovskikh, 2017-08-18:
Yes. The plugin 0.1 is incompatible with PyCharm 2017.2. Only with 2017.1. You can build the plugin from the master branch for 2017.2 as described in the readme. Don't hesitate to ask more questions if you encounter any issues.

closed 2017-08-18 by vlasovskikh