JetBrains/intellij-micropython #15: Can't run under Pycharm 2017.2

mitchins, 2017-09-05:
I'm kind of stuck what to do now.

The plugin isn't showing under 20172 because the published version is only listed as being compatible with 2017.1 I think.

So I downloaded the sources and tried to run using `./gradlew clean runIde`, but after downloading a whole bunch of things it just sits there and does nothing.

If there could be a little more elaboration on what should happen here it would help, as I am not sure if it is supposed to launch the Idea, or if I am supposed to copy something to a certain folder.

Either this or any other way to manually install/run with PyCharm 017.2 would be greatly appreciated.

vlasovskikh, 2017-09-05:
@mitchins Yes, I haven't published the plugin for 2017.2 yet. The current master branch is compatible with 2017.2, but there are few things I'd like to fix before announcing initial support for ESP8266 and Pyboard. Cloning the repo and running `./gradlew clean runIde` should download IntelliJ 2017.2, build the MicroPython plugin, and run IntelliJ 2017.2 with the plugin enabled. Do you see any error messages when you run this command? Which step does it stop on?

You can also use the same command to build and run the plugin with PyCharm if you uncomment `pycharmPath` variable in `gradle.settings` and specify full path to your PyCharm installation (the directory inside your installation that contains bin/, lib/, license/, and other internal PyCharm folders).

mitchins, 2017-09-05:
At first I used the project as-is.

After digging around I uncommented the pycharm path, ut since I'm on Mac OSX I pointed it directly to the binary e.g. /Applications/Pycharm/Contents/MacOs/pyCharm Does it need the root path or the binary directly?

vlasovskikh, 2017-09-05:
@mitchins The path is /Applications/PyCharm.app/Contents.

mitchins, 2017-09-06:
```
    …/~  ls -lh /Applications/PyCharm\ CE.app/Contents/{lib/pycharm.jar,MacOs/pycharm}                                                                                                                                                                0.01   11:29:51
-rwxr-xr-x@ 1 m044121  admin   139K 22 Aug 06:17 /Applications/PyCharm CE.app/Contents/MacOs/pycharm
-rw-r--r--@ 1 m044121  admin    98M 22 Aug 06:16 /Applications/PyCharm CE.app/Contents/lib/pycharm.jar
```

mitchins, 2017-09-06:
Did that, and it runs like a charm!
So to sum up. If you are on a Mac...

`gradle.properties`

`# Run MicroPython plugin inside PyCharm installed into the following path
pycharmPath=/Applications/PyCharm\ CE.app/Contents`

closed 2017-09-06 by mitchins