JetBrains/intellij-micropython #16: Type hints for pyb module specific to Pyboard

vlasovskikh, 2017-09-09:
It would be great to have good code analysis for the `pyb` module specific to Pyboard. We need a Python stub file similar to those ones in the typehints/ directory.

vlasovskikh, 2017-09-09:
@stefanhoelzl How do you feel about contributing the initial version of the type hints for the `pyb` module? It would enable good code completion / type checking / documentation for Pyboard in the MicroPython plugin.

domgiles, 2018-02-04:
It's not clear to me... Is anyone working on this?

vlasovskikh, 2018-02-07:
@domgiles No, there isn't anyone who is working on it. A pull request with `pyb` stubs similar to what we have in the typehints/ directory is welcome.

domgiles, 2018-02-13:
I'll see if I can find some time to take a look at it. No promises. I'm more comfortable in Java but I've a reasonable grasp of Python. It looks like more of a editing issue than anything really creative... right?

vlasovskikh, 2018-02-13:
@domgiles Yes, it is more about converting the docs for `pyb` into the Python stubs format, copying docstring and specifying types.

domgiles, 2018-02-15:
First let me start bu apologising for my lack of experience in working with plugins for the JetBrains Environment. I've pulled the code. Can you explain a couple of things. On running ./gradlew clean runIde it launches the intellij IDE (Which I already have the project in). I'm not sure but I was expecting to launch PyCharm with an updated build of the plugin installed. Did I miss something? i.e. Whats the workflow to enable me to make changes and then test them in PyCharm?

vlasovskikh, 2018-02-17:
@domgiles Yes, `./gradlew clean runIde` launches IntelliJ with the plugin by default, not PyCharm with the plugin. Please check the comments in `gradle.properties`. There you can uncomment the line with the path to PyCharm and specify the path to your PyCharm installation. It should be the path to the directory where `bin/`, `lib/`, `plugins/`, and other folders of PyCharm are located. In particular on macOS it's not /path/to/PyCharm.app, but /path/to/PyCharm.app/Contents/. PyCharm should be installed locally, `gradlew` won't download PyCharm for you.

If you're not familiar with IntelliJ, you can just edit files in your Git working directory and then launch PyCharm via `./gradlew runIde`. Alternatively, you can open the intellij-micropython project in IntelliJ Community Edition and edit code there. Make sure you've imported the project from existing sources as a Gradle project. You can then use "View | Tool Windows | Gradle" to launch Gradle tasks. IntelliJ will provide you with code highlighting and smart code completion for Kotlin and Python (download the Python plugin for the latter).

domgiles, 2018-03-04:
> , or mute the thread
> .<embedded image>

Sorry for the delay

Ok… All good now..

Started getting the pbyoard doc ported… It’ll be off and on but I can at least see the process now.

Regards

Dom

przemekgalazka, 2018-04-13:
@domgiles any news on this port?

domgiles, 2018-04-16:
I started the work.

Annoyingly I was asked to head to do some work in San Francisco… So I’ve been away for 2 weeks. back next week so I can restart then. It’s not complex. Just a lot of it for the Pyboard.

Dom  

Regards

Dominic Giles

hlovatt, 2020-06-15:
Is there a start on this available? Wouldn’t mind seeing how much work is still needed.

vlasovskikh, 2020-06-18:
@hlovatt It's similar to what we have for ESP8266. We have several collections of type hints in directories [here](https://github.com/vlasovskikh/intellij-micropython/tree/master/typehints):

* esp8266/
  * Modules available only on ESP8266
* micropython/
  * Modules that are the same on all MicroPython devices (including Pyboard, ESP8266, ESP32)
* stdlib/
  * MicroPython's micro-standard library, u-prefixed modules
* pyboard/ # Missing!
  * A place to put Pyboard specific stubs

Usually I write *.pyi files with type hints manually just by looking at the official API docs for the corresponding module and experimenting with it in REPL when I'm not sure about types, arguments, etc. There is a discussion about a tool that can generate stubs for MicroPython modules automatically, see #102 for that.

w0nd4bra, 2020-06-18:
@vlasovskikh I just checked the output of the stubber more precisely and there are some confusing stuff around. Some modules seem not to exist, while classes are stubbed as functions.
But I think it still saves work and time.
I just forked your rep to write the stubs, I'm bored.

hlovatt, 2020-08-10:
Have issued a couple of pull requests for the PyBoard Typesheds.

hlovatt, 2020-08-24:
Are you still interested in typesheds for PyBoard? I ask because I have a couple of pull requests in for a few weeks now.

I have more typesheds at:

https://github.com/hlovatt/PyBoardTypeshed

Plus there is a description, in README at same project, of how to 'monkey-patch' typesheds into the MicroPython plugin.

vlasovskikh, 2020-09-21:
Thanks to @hlovatt we now have a stub for `pyb` 🎉 Howard did a great job by writing the type hints for all the APIs of `pyb` manually and then putting all the docstrings in place using his [rst2pyi generator script](https://github.com/hlovatt/PyBoardTypeshedGenerator).

I'm closing this issue as done for now. If there is anything missing, feel free to open new issues or send your PRs with updates to stubs.

In a week I'll release a version of intellij-micropython with bundled `pyb` enabled for "Pyboard" device types and other changes. It will be compatible with PyCharm 2020.2.

closed 2020-09-21 by vlasovskikh