JetBrains/intellij-micropython #18: Typehints

Cediddi, 2017-09-13:
Hi, I'm writing stub files for micropython, especially for esp8266 port. I'd like you to share your thoughts with me, maybe we can collaborate on the future.

I've finished machine, Pin and I2C stubs for esp8266, but currently I can't find a way to solve the 'const' issue I'm having. 'const' is a function in 'micropython' library and is not present in locals(). In micropython, there's no '\_\_builtins\_\_' in locals() either.

If you don't import it, 'const' is not usable like a function, but you can still use it in a assignment situation. If you import it, it's usable like a function and the functionality is the same.

```
>>> const
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'const' is not defined
>>> const(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'const' is not defined
>>> x = const(1)
>>> from micropython import const
>>> const
<function>
>>> const(1)
1
>>> x = const(1)
>>>
```

Some libraries use it without importing and this gives a warning in pycharm. Do you think there's a way to inject this to __builtins__ as a stub? If I can walkaround this issue or if this plugin can handle the issue under the hood, It'd be great.

When I release the stubs, I'll link the repo here so you can check it. I omit documentation and return values but I'll add return values soon. I expect to finish most of the stubs before november.

vlasovskikh, 2017-10-20:
Hey @Cediddi, sorry for my late reply. Indeed, `const` seems to be a compiler directive, not a real function. Type hints is not the only way to provide information for code analysis in PyCharm and the Python plugin for IntelliJ. We can provide the resolve target for `const` via a custom `PyReferenceResolveProvider` or we can just ignore this name in the unresolved references inspection via a `PyInspectionExtension`. You can find these classes in the PyCharm code base. If you are not comforable with Java or Kotlin, I can do it myself.

vlasovskikh, 2017-10-20:
@Cediddi Just forgot to mention, I would appreciate pull requests with type hints even if they are incomplete yet.

vlasovskikh, 2017-12-09:
Hi @Cediddi! Any progress with your type hints for ESP 8266? I'd like to release a new version of the MicroPython plugin some time soon and extra type hints for ESP 8266 specific modules would be very welcome.

Do you need any help with `const`? Do you want me to implement it?

vlasovskikh, 2018-01-10:
Type hints for `machine.Pin`, `machine.I2C`, and many others were added in 7d8eedadddbf7bb4fa31ea4152a741b73d5d257a and subsequent commits.

closed 2018-01-10 by vlasovskikh