JetBrains/intellij-micropython #20: How to Install the Plugin with IntelliJ IDEA?

tobiashochguertel, 2017-11-01:
I can't find the plugin in the Repository / Plugin Manager:

[image]

Is there a special Repository URL which I need to add first?

Actpohomoc, 2017-11-01:
Same for me with PyCharm Pro and community editions

vlasovskikh, 2017-11-01:
@tobiashochguertel @Actpohomoc The published version of the plugin is compatible only with IntelliJ 2017.1. Sorry for the inconvenience.

I will publish a version for 2017.2 (?) and 2017.3 later. Meanwhile you can get the source code and run:

```
./gradlew buildPlugin
```
Then you can go to "Preferences | Plugins" menu, click "Install plugin from disk" and choose build/distributions/intellij-micropython-*.zip.

Did it help?

vlasovskikh, 2017-11-01:
@tobiashochguertel Thanks for the update!

closed 2017-11-01 by vlasovskikh