JetBrains/intellij-micropython #23: Is music module supports?

kylinpoet, 2017-12-04:
Then,how to use it?

vlasovskikh, 2017-12-06:
@kylinpoet Are you referring to the `music` module of BBC Micro:bit? We don't have a type hinting stub for this module yet. It means that the plugin won't be able to recognize methods from this module or provide code completion for it. [Here](https://github.com/vlasovskikh/intellij-micropython/tree/master/typehints/microbit) you can browse available type hints. A pull request with a stub for `music` would be welcome. The docs for the module are located [here](http://microbit-micropython.readthedocs.io/en/latest/tutorials/music.html).

vlasovskikh, 2018-01-10:
Fixed in d3d8ef68264d876e5f86af56335c71382cb148fd.

closed 2018-01-10 by vlasovskikh