JetBrains/intellij-micropython #28: Possible syntax errors for microbit typehints

MrYsLab, 2018-01-16:
The signature for uart.init
```
def init(baudrate: int = 9600, bits: int = 8, parity: int = None, stop: int = 1, 
*, tx: MicroBitDigitalPin = None, rx: MicroBitDigitalPin = None) -> None:
```
and 
display.show
```
def show(iterable: Iterable[Image, str], delay: int = 400, *,
wait: bool = True, loop: bool =False, clear: bool = False) -> None:
```
as well as display.scroll
```
def scroll(string: str, delay: int = 150, *, wait: bool = True,
loop: bool = False, monospace: bool = False) -> None:
```
all contain an asterisk or star as a parameter. This is consistent with the signature in the micro:bit MicroPython API, but I don't believe it is correct and they should be removed.

If you agree, I would be happy to issue a pull request and make the changes.

AugustH, 2018-01-16:
imho the star is ok, it is the marker for keyword-only parameters -- see
https://docs.python.org/3/glossary.html#term-parameter

vlasovskikh, 2018-01-16:
@MrYsLab As @AugustH mentioned, this is a keyword-only arguments separator introduced in Python 3.0. See https://www.python.org/dev/peps/pep-3102/

Feel free to re-open / comment on the issue if you think that the problem is still relevant.

closed 2018-01-16 by vlasovskikh

MrYsLab, 2018-01-17:
I guess my python 2.7 knowledge is showing and so is my lack of 3.6 knowledge ;-).  I have never come across this syntax before - live and learn.