JetBrains/intellij-micropython #29: Add microfs functionality for micro:bit persistent file system

MrYsLab, 2018-01-16:
When adding an external module, such as servo.py, it needs to be added to the micro:bit persistent file system. Having the 4 [microfs](https://github.com/ntoll/microfs) commands, ls, put, get, and remove available from within the plugin would be very helpful.

vlasovskikh, 2018-01-16:
For other MicroPython devices the plugin provides uploading services via a Run Configuration. You can upload either a folder or a file. When you right-click on a file.py and use "Flash file.py to device", the plugin creates a Run Configuration for you. There is also an action to clean the file system.

I think it's feasible to extend this Run Configuration for Micro:bit devices so that you can upload and delete files in a uniform way.

MrYsLab, 2018-01-18:
This may cause more issues for you than it is worth. One cannot use both the REPL and persistent file system manipulation at the same time - they both compete for use of the serial port.  This happens in the mu editor as well - if you open the REPL and press Files, a warning dialog box opens.

In addition, my reasoning for having the feature was to be able to load a 3rd party library into the persistent file system as per this [article](https://microbit-playground.co.uk/howto/add-python-module-microbit-micropython).

I am not sure if just copying and pasting the library to the top of the application source code is not a better idea. It is less pythonic, but I don't think importing the module saves any space. In addition, if you make changes to the application, you need to reload the library back into the persistent file system. Very annoying.

So what I am saying is, please feel free to close this issue. I will leave the choice up to you.

MrYsLab, 2018-01-18:
Also, it is very easy to add microfs as an external tool if one wishes to do so. https://mryslab.github.io/pseudo-microbit/install/#adding-uflash-and-microfs-as-external-tools

vlasovskikh, 2018-01-18:
Yes, adding these commands via external tools it a good workaround.

I'll keep this issue open as the problem with having both REPL and microfs commands running at the same time seems to me like a natural limitation that will be understood by the users.