JetBrains/intellij-micropython #32: micro:bit "Remove All Files From Device" not working

MrYsLab, 2018-01-18:
I tried select Tools/MicroPython/Remove All Files From Device and nothing was removed from either the persistent file system or the uploaded app.

I tried running adafruit-ampy locally and it does not seem to work with the micro:bit. When I try to run it I get a huge list of tracebacks.

I don't know if microfs works for the other boards that the plugin supports (I only own a micro:bit), if it does, perhaps it would make sense to swap it for the adafruit library.

vlasovskikh, 2018-01-18:
I use `uflash` to flash a file to a Micro:bit device in the plugin. Removing them will fail for the reasons you've mentioned. Yes, `microfs` could be a solution for Micro:bit in this case.

It looks like it will require extracting the logic for running `microcleanfs.py` from `RemoveAllFilesFromDeviceAction.kt` and putting it into `Esp8266DeviceProvider.kt`. A new API method in `MicroPythonDeviceProvider` is needed. It could be something like `getRemoveAllFilesCommand(facet: MicroPythonFacet): List<String>?`.

@MrYsLab Would you like to submit a pull request that introduces a `microfs`-based solution?

MrYsLab, 2018-01-18:
I am not sure how quickly I can get to this but would be willing to give it a try. That being said, without having the ability to send something to the file system from PyCharm, I am not sure if it makes much sense to be able to remove things from it for the micro:bit.

If this still makes sense to do, I would like to try is to add all of microfs functionality then, in spite of the REPL conflict. Can the drop down menu choices be local to the board type (esp8266 vs micro:bit) or is it a common set. If it is common, then forget this idea, since it will create work for the other boards.

I do not understand the plugin architecture very well, nor have I done any kotlin programming (but have done some Java), so I may not be the best candidate for the job, but would be willing to give it a go.

You mention Esp8266DeviceProvider.kt above. Did you mean MicroBitDeviceProvider.kt?

vlasovskikh, 2018-01-18:
What do you think about uploading files via a run configuration in the same way it's currently done for ESP8266 and Pyboard? When you specify the path to a directory, the plugin uploads the contents of the directory excluding any sub-directories marked as excluded.

I meant that the current `RemoveAllFilesFromDeviceAction` supports only ESP8266 so it's worth moving its logic into `Esp8266DeviceProvider` and change the action to just invoke a method of the `MicroPythonDeviceProvider` inferface.