JetBrains/intellij-micropython #33: Plugin fails to upload file to subdirectory

kaybee335, 2018-01-20:
Results in stack trace.  Strangely it actually seems to create the directory but then fails to upload the specified file into it.

Redacted Stack Trace tail:

```
`C:\Users\...\condaenvs\upython\pythonw.exe C:\Users\...\.PyCharm2017.3\config\plugins\intellij-micropython/scripts/microupload.py -C "C:/Users/.../test" -v COM4 "C:\Users\...\test1.py"
Connecting to COM4
Uploading files: 0% (0/1)
C:\Users\...\test1.py -> lib\test1.py
Traceback (most recent call last):
  File "C:\Users\...\.PyCharm2017.3\config\plugins\intellij-micropython/scripts/microupload.py", line 138, in <module>
    main(sys.argv[1:])
  File "C:\Users\...\.PyCharm2017.3\config\plugins\intellij-micropython/scripts/microupload.py", line 79, in main
    files.put(remote_path, fd.read())
  File "C:\Users\...\condaenvs\upython\lib\site-packages\ampy\files.py", line 137, in put
    self._pyboard.exec_("f = open('{0}', 'wb')".format(filename))
  File "C:\Users\...\condaenvs\upython\lib\site-packages\ampy\pyboard.py", line 265, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nOSError: [Errno 13] EACCES\r\n')
`
```

vlasovskikh, 2018-01-20:
Thanks for reporting this bug.

By the way, the upload log actually contains the Python command line the plugin is using for uploading the subdirectory:

```
C:\Users\...\condaenvs\upython\pythonw.exe C:\Users\...\.PyCharm2017.3\config\plugins\intellij-micropython/scripts/microupload.py -C "C:/Users/.../test" -v COM4 "C:\Users\...\test1.py"
```

How do you feel about investigating this problem and sending a pull request? It would be very helpful if you tried debugging this problem in PyCharm by creating a Python run configuration with these command line parameters and debugging it.

kaybee335, 2018-01-25:
Sorry @vlasovskikh.  I got pulled away by a couple of other things.  I will get back to looking at this today as well my REPL issues.  Any further progress/issues with the REPL I will post as a new Issue as they seem to be specific to the way I have things set up.

kaybee335, 2018-01-25:
Please Close.  This seems to be related to my other issue concerning python.exe/pythonw.exe:

[Bug #37](https://github.com/vlasovskikh/intellij-micropython/issues/37)

An upload to a subdirectory works properly for me now if I use python.exe rather than pythonw.exe.  Directories are also created as required.

closed 2018-01-28 by vlasovskikh

Mendez83, 2018-05-22:
Sorry to come back, but I have the same issue when try to upload a subfolder content.

`Connecting to COM3
Uploading files: 0% (0/5)
C:\Users\XYZ\Pycharm-ESP32\cls_parameters.py -> cls_parameters.py
Uploading files: 20% (1/5)
C:\Users\XYZ\Pycharm-ESP32\main.py -> main.py
Uploading files: 40% (2/5)
C:\Users\XYZ\Pycharm-ESP32\classes\cls_parameters.py -> classes\cls_parameters.py
Uploading files: 60% (3/5)
C:\Users\XYZ\Pycharm-ESP32\configs\net.json -> configs\net.json
Traceback (most recent call last):
  File "C:\Users\XYZ\.PyCharm2017.2\config\plugins\intellij-micropython/scripts/microupload.py", line 138, in <module>
    main(sys.argv[1:])
  File "C:\Users\XYZ\.PyCharm2017.2\config\plugins\intellij-micropython/scripts/microupload.py", line 79, in main
    files.put(remote_path, fd.read())
  File "C:\Program Files\Python36\lib\site-packages\ampy\files.py", line 137, in put
    self._pyboard.exec_("f = open('{0}', 'wb')".format(filename))
  File "C:\Program Files\Python36\lib\site-packages\ampy\pyboard.py", line 265, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nOSError: [Errno 13] EACCES\r\n')`

Seem that for the first file in subfolder it's work.

I'am on Pycharm 2017.2.3 and try to run on latest firmware esp32-20180522-v1.9.4-67-gcda96419.

vlasovskikh, 2018-05-22:
@Mendez83 Please try to debug your uploading problem by running C:\Users\XYZ\.PyCharm2017.2\config\plugins\intellij-micropython/scripts/microupload.py from a terminal with appropriate parameters. Its source code could help.

jeromefischer, 2019-10-12:
I run in the same error: 
`Connecting to COM4
Uploading files: 0% (0/6)
D:\ESP32\uPytonExample\main.py -> main.py
Uploading files: 16% (1/6)
D:\ESP32\uPytonExample\.idea\misc.xml -> .idea\misc.xml
Uploading files: 33% (2/6)
D:\ESP32\uPytonExample\.idea\modules.xml -> .idea\modules.xml
Uploading files: 50% (3/6)
D:\ESP32\uPytonExample\.idea\uPytonExample.iml -> .idea\uPytonExample.iml
Traceback (most recent call last):
  File "C:\Users\name\.PyCharmCE2018.2\config\plugins\intellij-micropython/scripts/microupload.py", line 138, in <module>
    main(sys.argv[1:])
  File "C:\Users\name\.PyCharmCE2018.2\config\plugins\intellij-micropython/scripts/microupload.py", line 79, in main
    files.put(remote_path, fd.read())
  File "D:\ESP32\uPytonExample\venv\lib\site-packages\ampy\files.py", line 209, in put
    self._pyboard.exec_("f = open('{0}', 'wb')".format(filename))
  File "D:\ESP32\uPytonExample\venv\lib\site-packages\ampy\pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1\r\nSyntaxError: invalid syntax\r\n')
`
The problem was, that in PyCharm I set the path for the "Run/Debug Configuration" to the directory where my PyCharm project is located (e.g. D:\ESP32\uPytonExample). But actually the **Path** has to be set to the file that has to be uploaded: **D:\ESP32\uPytonExample\main.py.** 

Then it works! So it is not a bug or issue, it is only a wrong configuration of the