JetBrains/intellij-micropython #34: Feature Request: Edit Configurations 'View All Files' checkbox

kaybee335, 2018-01-20:
Currently the Edit Configurations file browser seems to filter so that it only displays python files in the project.  I sometimes keep configuration information in JSON files which have a '.json' extension.  I now have to go to another tool (ampy) to upload these files or mess with the configuration editor 'typing blind' (it will actually accept a file name like config.json is you type it in with the appropriate dir AND also upload it to the board).

If there was a "View All Files" checkbox on the main Edit Configurations window that refreshed the file browser appropriately before displaying it would be consistent with other apps that generally take ths type of approach.

vlasovskikh, 2018-01-20:
Thanks for filing this. I've never encountered it personally since the run configuration allows you to specify the path to a directory and uploading the whole project (with some sub-directories marked as excluded via their context menu) was usually enough for me.

The plugin should definitely allow uploading a single non-Python file. I'm not sure about creating a run configuration via right click on a file by default as we do for Python files. @kaybee335  What do you think?

kaybee335, 2018-01-26:
Actually it does work if you set up your Project Structure to exclude .idea directory.  If you then upload the entire project by specifying the root directory of your project, it will upload everything no matter what the filename is.  The Exclude Files entry seems to be ignored, not sure if this is by PyCharm or the plugin.  Annoyingly the PyCharm Project Structure dialog does not show you individual files to include/exclude, I guess that is why it is called "Structure".

I also didn't realise that I could right click on a python file to automatically create the run configuration!!  That's sweet, I've been doing it manually by copying other configurations up to now.

As far as the suggestion, it would certainly be consistent if you could right click to create a run configuration.  I don't have any idea of the complexity of implementing it.  

I think i would be happy if I could just select it as the Path Target in the Run/Debug Configurations dialog.  It would do what I need if that dialog had a Show All Files checkbox that then allowed selecting a JSON (or other) file rather the file list being restricted to python files as it is now. Of course that's likely even more difficult to implement than the right click....  :-0

kaybee335, 2018-01-29:
Updated README.md in push #38 to show how to upload/flash non-python files.

vlasovskikh, 2018-05-30:
Closing this one since it's already possible and documented in the README.

closed 2018-05-30 by vlasovskikh