JetBrains/intellij-micropython #35: Feature Request: Opton to preserve boot.py when Removing All Files from Device

kaybee335, 2018-01-20:
Currently the Remove All Files from Device option also removes boot.py.

This is not a file you typically change in development (that is usually main.py and any others you create).  That is why it is actually provided as part of the micropython firmware image.

I would like to see a checkbox allowing boot.py to be preserved during an Erase.  In fact it might be better implemented the other way around, you actually have to select the checkbox in order to erase boot.py as part of the device erasure, that will at least make users think about it.

vlasovskikh, 2018-01-20:
Yes, it makes perfect sense NOT to delete boot.py. I propose to make this a non-configurable default for now and see if anyone really need this option.

kaybee335, 2018-01-25:
I will grab this one.  I have had a quick look at the code and think I can do it pretty quickly.  I propose adding an argument to microcleanfs (something like -b).  If the flag is not specified, then boot.py is preserved.  If the flag is specified then boot.py will be removed with the other files.  This should make it fairly easy to hook into the Micropython Languages and Frameworks page if desired.

I'll get the changes working independently and then check with you to see if you like the approach.

kaybee335, 2018-01-26:
Okay, got it.

Changed microcleanfs.py to take a (-f --force) option which will delete ALL files.  If the option is not specified then any filenames in the list exclude_files[] are NOT deleted.  exclude_files[] currently only contains boot.py but I tested it with multiple files (works), top level directories (works) and subdirectories (do not work because they get deleted when the parent is deleted unless the parent is also specified as an exclude).

Nothing I changed should be platform specific.  Now I just have to figure out how to submit my changes... I am GitHub virgin so it might be smart to go watch a tutorial or two before I blow up the entire plugin project :-)

kaybee335, 2018-01-26:
[Pull #38](https://github.com/vlasovskikh/intellij-micropython/pull/38)

vlasovskikh, 2018-02-10:
Pushed to master.

closed 2018-02-10 by vlasovskikh