JetBrains/intellij-micropython #36: ESP32 support please?

idesignstuff, 2018-01-20:
We all love Micropython on the ESP8266 until it runs out of memory.
The ESP32 offers a mot more memory, dual cores, as well as BLE and more IO, including several ADC pins.

vlasovskikh, 2018-01-20:
Even though the MicroPython plugin for PyCharm doesn't officially support ESP32 yet (I haven't got any ESP32 devices yet, so I cannot test it and fix potential issues), I've seen several people on Twitter reporting that they've successfully used the plugin with ESP32 devices. I presume they used ESP8266 as the board type in the settings.

@idesignstuff Could you please try setting ESP8266 as the device type and report any problems or features I think are necessary?

idesignstuff, 2018-01-20:
will do! I just assumed it wouldn't be that compatible.

vlasovskikh, 2018-01-20:
See also this Twitter thread https://twitter.com/kaybee335/status/953412449931182081

idesignstuff, 2018-01-21:
So, on win10, at least a few things aren't right for me here. 
I've followed the apparently really clear step-by-step here:
https://blog.jetbrains.com/pycharm/2018/01/micropython-plugin-for-pycharm/

A couple of things are unclear to me, especially the "path to device."
On win10, it should be, for example, COM3, but not sure if I need //, \\, /, or \ before and or after the COM3
is it \\COM3
\COM3
\\COM3\
I don't seem to need any extra characters to use PuTTY. It's just COM3

I've got Python3.6.x, and PyCharm 2017.3
I'm a beginner with PyCharm, a beginner with MicroPython, and a beginner with the MicroPython plugin.
It's probably me.

vlasovskikh, 2018-01-21:
@idesignstuff When I was writing this blog post I was still unable to test the plugin on Windows since I work on macOS and Linux exclusively. Since then I borrowed a Windows machine and fixes the issue with connecting to devices via USB on Windows.

Indeed, the names of the devices on Windows are not paths but rather names: COM1, COM2, etc. In the MicroPython plugin you can use these names without any path components or slashes.

Regarding the issue with run, it seems the MicroPython support hasn't been activated in the settings for your project or the plugin has failed to generate a MicroPython run configuration on right-click on the file. Try to do it manually. Go to "Run | Edit Configurations..." menu and create a new "MicroPython" run configuration. Specify a path to your Python file there. If you don't see "MicroPython" in available run configuration types then it's likely the MicroPython support is not activated. See "File | Settings | Languages & Frameworks | MicroPython".

idesignstuff, 2018-01-21:
It looks like Python 3.5 is key.
I've installed that, and the plugin seems to be working more like described
in your intro.

Christian

idesignstuff, 2018-01-24:
C:\Users\[redacted]\PycharmProjects\clock\venv\Scripts\python.exe
C:/Users/[redacted]/PycharmProjects/clock/venv/main.py
Traceback (most recent call last):
  File "C:/Users/[redacted]/PycharmProjects/clock/venv/main.py", line 2, in
<module>
    import network
ImportError: No module named 'network'

Process finished with exit code 1

Christian

vlasovskikh, 2018-01-24:
@ideaignstuff It looks like you’re using a loca Python rum configuration instead of a MicroPyhton run configuration for flashing files to devices. See “Run | Edit Configurations” menu and the README.md on GitHub.

kaybee335, 2018-01-25:
I can confirm the plugin works with an Adafruit ESP32 Feather Huzzah.  I do have some issues with getting a REPL (which may be ESP32 specific now that I think of it, there are quite a few differences in the base micropython libraries (8266 - no _threads, ESP32 has _threads as an example).  I will try with an ESP8266 as part of my issue isolation efforts.

If you are using your computers system python then make sure it is 3.5 (this is where IntelliSense goes for anything it doesn't 'understand').   I strongly recommend using a virtual environment so that you can manage the version of python that the plugin ends up using for the non-micropython aspects.  Don't use 3.6 if you can avoid it, otherwise new syntactic elements such as f-strings will not be flagged as errors in the editor which they certainly will be in your micropython runtime (I know, I've tried :-) ).  Technically, micropython seems to be more accurately described as 3.4 with a lot of 3.5 features added as far as the pure python aspects.

kaybee335, 2018-02-03:
@idesignstuff - is this still an issue for you?

idesignstuff, 2018-02-03:
I'll try again today. I *think* I got the plug-in to work once, so it may
be good old user error in configuring pycharm and venv.

Christian

kaybee335, 2018-02-03:
Okay @idesignstuff, keep us posted in this thread and we'll try and help out where we can.  I use conda environments rather than venv's but I do have everything working for an ESP32 Adafruit Feather Huzzah board including REPL support under a conda environment with just Python 3.5, docopt 0.6.2, adafruit-ampy 1.0.3 and esptool 2.2 (although I don't believe esptool is relevant).

idesignstuff, 2018-02-03:
That's encouraging. I wonder also if my direct from China modules are of
the best quality as well. I have 3 types.

Christian

idesignstuff, 2018-02-03:
> If you can communicate with the micropython devices using adafruit-ampy
> then the plugin should work (or at least be made to work).

I can confirm that flash and repl are working on one venv, using two
different makes of ESP32 board on a win10 x64 machine.
PEBKAC strikes again.

Christian

idesignstuff, 2018-02-03:
the detect functionality would be really helpful to know what COM port
these things show up on.

Christian

On Sat, Feb 3, 2018 at 4:14 PM, Christian Ritchie <we.eat.spamalot@gmail.com

closed 2018-02-03 by idesignstuff

kaybee335, 2018-02-04:
@idesignstuff - glad you  got things going! Enjoy!

The port scan does work for known USB/UART devices but I think that you may be caught by your clone boards this time.  I suspect that they are using a USB/UART chip that is not recognised by the port scan.  Both of my Adafruit boards (Feather ESP8266 and Feather ESP32) use the Silicon Labs CP210x and it is recognised by the scan.

I believe you are using Windows from your other issue about backspace in the REPL.  Open `Device Manager` and then select the connected USB/UART under Ports (COM &LPT), right click, choose `Properties` and then choose the `Detail` tab.  There is a dropdown on that tab, select `Hardware IDs` and you should get a display similar to the attached image:

[image]

The key information is displayed on the second line.  The two hex values shown `10C4` and `EA60` identify the vendor_id and port_id of the USB/UART and it is those two values that are needed to find the USB/UART in the scan.

If you would like to see your USB/UART supported by the scan, please open a new issue that provides those two hex values and as much information about the micropython board and/or USB/UART being used.  Information such as name, model, product web link, whatever you have would be useful.  In particular we need to know the target micropython architecture (ESP8266/ESP32/MicroBit/Pyboard) as each board architecture has a different set of IDs that are scanned in the current implementation of the plugin.

kaybee335, 2018-02-04:
I have created a script that performs the scan and returns the USB vendor_id:port_id for connected micropython boards.  This would be meant as a diagnostic utility used to provide accurate information about the various USB/UART devices that people are using.  I have tested it under Win10 and it works.  It does not work under Windows Subsystem for Linux (WSL) as the vendor_id:port_id information seems to be stripped out by the WSL serial port abstraction layer (all 100 serial ports in WSL report vendor_id:port_id as None:None so I don't think there is anything that can be done about this).

I will submit the utility script for review by @vlasovskikh as I cannot verify whether it works properly on linux or MacOS.

Just for fun, here is the output from the script on my system with four different boards attached (Feather ESP8266, Feather ESP32, microBit and WiPy 2.0).  As you may notice, Win has assigned two devices to COM3, not sure what is going on there, I seem to be talking to one of the boards via ampy and it seems to be the first COM3 listed).

[image]

idesignstuff, 2018-02-24:
[image: Inline image 1]

It's an ESP32 board with an integrated OLED and the Weoms logo on the back,
but pretty sure it's a knockoff.
It was a gift from my amazon list, from this link:  http://a.co/aSfS3lb

I've been able to get it to connect automatically to my home wifi, and
display status messages on the oled, but the process of uploading, checking
and communicating with it via usb/ampy/puTTY/pycharm has been *almost*
enough to make me want to quit.

Christian