JetBrains/intellij-micropython #37: Documentation Update - Win10 specific instructions

kaybee335, 2018-01-25:
Well, after chasing my REPL problem all day I found out what it was. A Problem Between Keyboard and Chair!

Windows installations have two python interpreters available: python.exe and pythonw.exe
I had always been selecting pythonw.exe as the project interpreter as the difference is whether or not it opens a new console window to execute and pythonw.exe does NOT which has never bitten me before.  It turns out that if you don't open a console, stdin, stdout and stderr are /dev/null'd rather than re-directed to the Miniterm instance in microrepl.py.  The solution is to use the python.exe executable and all is good on both the ESP8266 and ESP32 boards I tested it on (both AdaFruit Feathers).

Could you please add an addendum to the README.md for Windows users to make sure they select python.exe as the project interpreter rather than pythonw.exe?

kaybee335, 2018-01-26:
I have made the required change to README.md and also added two sections (using Project Structure to exclude .idea directory and creating a run config using a right click - sorry, my screenshots are light vs. your dark ones...).

I have one outstanding pull request for the boot.py issue and it appears I cannot make another for the doc change.  It didn't seem to make sense to combine them as they are two completely different topics.  What is the right way forward here please?

kaybee335, 2018-01-26:
Trying to figure it out and I seem to have combined this update with the boot.py one in Pull #38

vlasovskikh, 2018-02-02:
@kaybee335 I've assigned this issue to you since you've already submitted a PR that will fix it. Are you OK with that?

kaybee335, 2018-02-02:
Absolutely.

kaybee335, 2018-02-03:
The push addressing #38 should also address this in the updates to README.md.

vlasovskikh, 2018-02-10:
Pushed to master.

closed 2018-02-10 by vlasovskikh