JetBrains/intellij-micropython #39: microrepl.py, findusb.py - missing copyright headers, not using docopt for arg parsing

kaybee335, 2018-01-29:
The referenced files do not have copyright headers like microcleanfs.py and microupload.py.  Is the same header applicable to these two files?  

These two files also do brute strength arg parsing whereas the other two use docopt.  I think they all should use docopt to be consistent and also allow facilitate adding additional arguments at a future date.

I would be happy to make these changes if I know the answer to the copyright question (which I am pretty sure is yes but just want to confirm).

vlasovskikh, 2018-02-02:
microrepl.py should have its own copyright header, see licenses/microrepl_license.txt. microcleanfs.py and microupload.py should have our default copyright header.

Yes, I agree that switching to `docopt` will be more consistent. The only thing is findusb.py is used during device detection so one should check if error messages about not having `docopt` installed yet are not confusing to new users.

kaybee335, 2018-02-02:
@vlasovskikh - Okay, I have a branch that I am testing with the changes made.  I will check the behaviour when `docopt` is not installed in the python environment.  The user was going to hit the problem at some point anyway though as `microupload.py` and `microcleanfs.py` were already relying on `docopt`.

kaybee335, 2018-02-02:
What do you think should be returned by `findusb.py` via the print() statement if `docopt` is not installed? Currently it prints the matched device(s) if found or nothing if a device is not matched.  Would printing 'ERR' or something like that be appropriate?  I gather this is picked up by the `kotlin` side of things in order to display the detected device in the `Device path:`

vlasovskikh, 2018-02-02:
@kaybee335 The Kotlin function that runs this script is [MicroPythonFacet .findSerialPorts](https://github.com/vlasovskikh/intellij-micropython/blob/master/src/main/kotlin/com/jetbrains/micropython/settings/MicroPythonFacet.kt#L98). It actually assumes nothing was found if the script returns a non-zero code.

I guess it's fine to do just nothing: let an `ImportError` propagate so that the script returns a non-zero code. The only thing that is required is a bit of testing, so that a user without `docopt` is not actually affected when they configure their device.

After that when the user is about to run the REPL or flash some files the plugin shows the yellow warning on top of the editor that some packages are not installed. And I believe REPL and Run actions are disabled without installed dependencies.

kaybee335, 2018-02-02:
Okay, I'll poke at it and make sure things are as clean as possible if `docopt` isn't installed.  Easy enough to set up another conda environment without it to do the testing.

kaybee335, 2018-02-03:
Copyright headers have been included in #38 by accident.  I will submit the docopt changes as a separate push.

vlasovskikh, 2018-05-30:
I've updated the title, since the copyright headers are already there. The missing bit is parsing of the command line arguments via `docopt`.

@kaybee335 Are you still interested in contributing it? :)

vlasovskikh, 2018-07-25:
Closing this one since there has been no activity for a while.

closed 2018-07-25 by vlasovskikh