JetBrains/intellij-micropython #41: Can't flash ESP8266. Error: Module for path is not found

tarakanov, 2018-02-02:
I cannot flash my ESP8266 board. I have ESP-07 module (not WEMOS D1 mini nor WEMOS D1 mini Pro). I successfully flash it with other IDEs - Arduino and esptool.py. 
When I edit Run/Debug Configuration and select my USB/UART converter, it throw an error - `Error: Module for path is not found`, see screenshot. Maybe it is because I have external usb/uart converter. I have latest Pycharm and Micropython plugin and Ubuntu 16.04 lts

[image]

kaybee335, 2018-02-02:
In the screenshot you show:

**Name:** is what you wish to call the Configuration you are creatingeditong so you can find it in the dropdown later.
**Path:** is the path to the python file you wish to upload to the device..

The serial port and device type are selected under <em>File | Settings... | Languages and Frameworks | MicroPython</em>.  It is on that screen that you select the device type and specify the path to the serial port to be used.  This needs to be done before you can upload the file to the board but only needs to be performed once for the project.  The setup steps for a project are:

1 Create project and define python runtime (3.5 or newer - see notes in README about versions newer than 3.5)
2. Enable the MircoPython framework for the project - this defines the device type and serial device
path, make sure to also check the <em>Enable MicroPython Support</em> checkbox.
3.  Edit your Project Structure to exclude the .idea directory created by PyCharm.

You should be ready to start uploading files to the device now.

vlasovskikh, 2018-02-02:
Yes, the description by @kaybee335 is accurate, thanks @kaybee335 for handling it.

@tarakanov The name 'Path' could be confusing. Which names for this field you think would be more appropriate (but not too long)?

tarakanov, 2018-02-02:
Ok, now I understand. Thank you. 
But I face next problem. To upload to ESP8266 you have to manipulate two pins - RST and GPIO0. I used to connect them to USB/UART converter: ESP_RST to USB_UART_RTS and ESP_GPIO0 to USB_UART_DTR. `esptool.py` drive them as needed - both pull down in startup for a moment.
But micropython pulls RST and GPIO0 down constantly or waiting for an answer forever.  
```
/home/semyon/PycharmProjects/micropython-blink/venv/bin/python /root/.PyCharm2017.3/config/plugins/intellij-micropython/scripts/microupload.py -C /home/semyon/PycharmProjects/micropython-blink -v /dev/ttyUSB0 /home/semyon/PycharmProjects/micropython-blink/blink.py
Connecting to /dev/ttyUSB0
Uploading files: 0% (0/1)
/home/semyon/PycharmProjects/micropython-blink/blink.py -> blink.py
```

kaybee335, 2018-02-02:
`esptool.py` is generally only used to flash MicroPython itself on to the development device.  Flashing MicroPython to the device is not supported by the plugin.

What the plugin does is more analogous to what adafruit-ampy does: it copies python files to the device, erases files from the device, and things like that.  The only reset performed is a software reset equivalent to the keyboard initiated device reset entered at the device REPL.

As a bug isolation check, will [adafruit-ampy](https://github.com/adafruit/ampy) transfer files to the device without those lines (RTS/DTR) connected? The procedure used by the plugin is very similar.

kaybee335, 2018-02-02:
No problem @vlasovskikh, glad to help out as long as I am doing that and not confusing things :-)

I'll take a look at the README.md update that you are reviewing and see if I can do something to make the setup steps a little clearer.  Possibly just the high level overview steps as I outlined earlier in this thread will be enough.

vlasovskikh, 2018-02-02:
@tarakanov The MicroPython plugin for PyCharm uses [microupload.py](https://github.com/vlasovskikh/intellij-micropython/blob/master/scripts/microupload.py) to upload files to ESP8266 devices. It's a thin wrapper around the `adafruit-ampy` package you can find on PyPI. I guess that the second issue you've mentioned is an upstream problem. Could you please try to run `microupload.py` from the command line and debug what goes wrong in your case?

Does that answer your question?

tarakanov, 2018-02-03:
Oh, I was totally confused with micropython. I have no idea what micropython is and supposed [your tutorial](https://blog.jetbrains.com/pycharm/2018/01/micropython-plugin-for-pycharm/) was only instruction I need.  I read more about it and now understand the concept.
I followed [this tutorial](https://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/intro.html) and flash Micropython firmware - the base for any Micropython application. Then I came back to Pycharm and Run your example. And it run! But I had to detach RTS/DTR lines for this as they reset chip constantly. I believe it would be convenient not to drive them by `microupload.py`. I do not try `adafruit-ampy` by the moment.
I guess the issue is resolved. Thank you.

closed 2018-02-03 by tarakanov

kaybee335, 2018-02-03:
@tarakanov - I have taken an action to update the documentation to make it clearer that the plugin does not support flashing the micropython firmware image onto the device, just micropython program files.

I am still a little curious about the RTS/DTR lines.  The microupload.py code only specifies the baud rate when opening the port so all the defaults are used which is full software handshaking and no use of the hardware lines.  It is possible that they are being toggled anyway during the upload but I have no way of verifying that with the external USB/UART I have as it does not implement those lines.  I assume you are using a USB/UART similar to this one?

https://www.aliexpress.com/item/FT232RL-3-3V-5-5V-FTDI-USB-to-TTL-Serial-Adapter-Module-Mini-Port/32554612690.html

tarakanov, 2018-02-05:
@kaybee335 yes, I have similar adapter. I try to use `ampy` and it gives the same result - RTS goes down and prevent ESP to work. Even when I open serial connection with `picocom /dev/ttyUSB0 -b 115200`. I try all combination of reset/init/handshake with no success. 
But I am sure that RTS/DTR pins can be controlled by the adapter's driver. I am ready to take a closer look on this feature.  I have made several boards with this breakouts - RX/TX, RTS/DTR, VCC/GND. And it is very handy to burn firmware, you do not need to have additional switches or buttons for flashing.

kaybee335, 2018-02-05:
@tarakanov - the fact that none of the utilities you mention work with those lines connected indicates that this is not a plugin issue but something deeper in the adfruit-ampy/PySerial layers that the plugin relies on.  I am still a bit confused about why you need those lines connected.  The typical life cycle of a micropython application is:

1. Flash micropython firmware image to development board
2. Edit micropython application code in code editor
3. Upload micropython application code to dev board
4. Execute micropython application on dev board (likely after connecting a terminal and maybe using REPL)
5 Repeat Steps 2-4 until application works or lose interest/get distracted....

The plugin brings Steps 2-5 into PyCharm.  It also adds the benefit of IntelliSense in Step 2 so you get things like typehinting and knowledge of the core micropython libraries so the chances of your code executing correctly in Step 4 are vastly improved.

The serial line manipulation you seem to be referring to has only been required/relevant for me in Step 1 when the micropython firmware image is initially flashed to the device.  This is not currently supported by the plugin.  I have found no need with any of my boards to have those 'programming' lines connected for developing a micropython application once the micropython image is flashed on the development board.

Are you sure really need these lines to develop a micropython application (Steps 2-5) rather than flash the micropython firmware image to the board (Step 1)?

tarakanov, 2018-02-05:
@kaybee335  I agree with your arguments. Like I said I am new to Micropython and its new method to upload application. Before that (in case of esp-open-sdk or Arduino) you need to burn firmware completely every iteration. I just need to used to that thing. 

Today I try one more thing in micropython - freeze modules, when trying to implement websockets from [here](https://github.com/danni/uwebsockets). In which case you need to burn firmware. For that I try vagrant following [this tutorial](https://learn.adafruit.com/building-and-running-micropython-on-the-esp8266?view=all#overview). Vagrant is also new for me,  I successfully use it in console. I saw that pycharm has a vagrant feature. But I haven't try it yet. Maybe it will complete Steps with

Step 4.5. Decide to make or not freeze modules. If so - make new firmware via vagrant tool and upload it via esptool.

In that case you need to manipulate RST/GPIO0 pins.

kaybee335, 2018-02-05:
It was my understanding that when you create frozen modules or bytecode you are actually 'baking' these modules into the micropython firmware which is then put on the board in a repeat of Step 1.  There is no cycle of steps 4 and 5 because you have to rebuild the complete micropython firmware image from the full source with the appropriate toolchain in order to freeze modules and get them on the device.