JetBrains/intellij-micropython #42: Wireless Programming for micropython

wasanka, 2018-02-02:
It would be great to have wireless programming for micro controllers through web socket..

kaybee335, 2018-02-02:
@wasanka - I assume you mean in a manner similar to that provided by the webREPL but integrated with the plugin?  That may turn it into a PyCharm Pro only feature as it might be considered a 'remote development' system.  Maybe @vlasovskikh has some insight?

vlasovskikh, 2018-02-02:
Its a reasonable feature for this plugin.

kaybee335, 2018-02-02:
Okay, just figured it was worth asking the question before possibly heading full speed at a dead end :-)

vlasovskikh, 2018-02-02:
@kaybee335 Oh, are you actually interested in implementing it? :)

kaybee335, 2018-02-02:
@vlasovskikh - I find it intriguing.  My recollection of using the webREPL was that it was convenient but also quite flakey but that was quite a few versions of micropython ago.  I also believe the webREPL is pretty big in resource footprint so most people don't usually enable it as it severely limits the resources available to the application.

The thought of updating the application on my flock of micropython devices without having to make a physical connection is very appealing.  I would certainly be willing to spend a bit of time to investigate feasibility.  I have also said webREPL from the outset, there may be a better way, I'll have to do some dredging through the various forums to see what is going on on that area as I know it is a hot topic.

As another caveat, I don't believe all devices have webREPL, I seem to recall that the ESP32 doesn't.

wasanka, 2018-02-03:
Yes, of cause. Because once we flash the MCU then either we can work with COM ports or wireless programming. Most of the IOT development benefits with wireless programming because once we program the chip and plug it in to a device then its difficult to connect back to a computer each time when we need to update to fix a bug. Wireless programming makes it super easy.

webREPL already helps us to upload/download files and we can use commands to create directories, list file system etc.

wasanka, 2018-02-03:
If we can have something like a view on the side (e.g. project explorer, ant build etc) to connect to the device with ip/port and then list the file system, double click to open the file from device in an editor etc....

wasanka, 2018-02-09:
Check this out https://github.com/BetaRavener/uPyLoader

vlasovskikh, 2018-02-10:
@wasanka Thanks for sharing this link.

kaybee335, 2018-02-11:
@wasanka - thanks for the link.  From what I can tell, that project is using the webREPL as the 'transport' layer which limits the number of board types it will work on.  I am still checking into this, it isn't forgotten but I am having to go a little deeper into things than I initially thought.