JetBrains/intellij-micropython #43: repl keyboard not sending right keys for US on Win10

idesignstuff, 2018-02-03:
When using the Repl in pycharm, my backspace key sends a "[?[K" and if I hit backspace again the "K" is replaced by "?[K"
If I hit the delete key, the repl prints an "S"

In the IDE text editor inside of PyCharm, this is not an issue.
Using PuTTY, this is not an issue.

Is this a config setting in the plugin, or just a localization error?

kaybee335, 2018-02-04:
@idesignstuff - I have the same issue under Win10.  Is that the OS you are running as well?  I am making that assumption due to your mention of `PuTTY`.

If you are running under Win10, then the backspace does actually work, it just displays junk when you press it.  If you ignore the 'line noise' displayed you can proceed with your REPL interaction. 

Terminal handling tends to be very OS specific, especially with special characters like `Backspace`.  It is definitely something about the `Backspace` handling and not the key itself as `Ctrl-H` also has the same behaviour.

Can you pleasee confirm that you are running Win10?  I would also like confirmation from @vlasovskikh that Backspace works as expected in Linux/MacOS.  That should validate it as a Win10 only issue and if that is the case I will take this issue on and see what I can do about it.

idesignstuff, 2018-02-04:
@kaybee335, that's right, I'm using win 10, creator's fall update.

Christian

kaybee335, 2018-02-04:
@idesignstuff - Okay, that is the same as I am using so our experiences with the plugin should be similar.

QiweiPy, 2018-05-29:
First of all, thanks to Andrey Vlasovskikh, I in the windows10  system, using the pycharm2018 version, starting Micropython REPL, in the input box, the backspace key and the direction key display chaotic code, almost not normal use.Is there a solution to the problem at present?

vlasovskikh, 2018-05-30:
@QiweiPy Does your MicroPython REPL behave as expected when you run it from the terminal? The plugin uses [this script](https://github.com/vlasovskikh/intellij-micropython/blob/master/scripts/microrepl.py) to run REPL. The problem is likely either in this script (try it in the terminal and check if it works) or the Java terminal library the IntelliJ platform uses for the terminal tool window.

QiweiPy, 2018-06-02:
@vlasovskikh 
I use putty and other similar software tools to work properly. When I run the microrepl.py script at the windows10 command prompt window, it is still the case, and because of my limited knowledge, I don't know how to deal with this situation.

Streyder, 2018-06-21:
This is also happening on a english W10 using both En-Us and De-De Keyboards

JohnMTu, 2018-10-01:
similar issue on w7 with US keyboard

junhuanchen, 2018-10-24:
This problem has been repaired on my mpfshell.
And I did a demonstration and ported a new plug-in version.
https://github.com/junhuanchen/intellij-mpfshell

Kortenbach, 2018-12-15:
US keyboard Win 10. Same problem.

Theeky, 2019-01-06:
Same problem in windows 10 , portuguese keyboard!

peterschlem, 2019-01-10:
DITTO Windows 10 / US Keyboard - same problem with microrepl.py . 
I am guessing the intellij-mpfshell will install as a zip????

junhuanchen, 2019-01-11:
> DITTO Windows 10 / US Keyboard - same problem with microrepl.py .
> I am guessing the intellij-mpfshell will install as a zip????
> 
> Did not install as a plugin error: Failed to load plugin descriptor .....

You can use zip directly [intellij-mpfshell](https://github.com/BPI-STEAM/BPI-BIT-MicroPython/releases/tag/intellij-mpfshell-1.1.1), try it tested.

peterschlem, 2019-01-11:
Juwan,
The recommended link installs successfully however mpfshell findusb.py fails - looks like comports() does not match the USB port ids whereas the same code in the Micropython plugin works for the esp8266 settings.
Target is an esp32.

Peter Schlemowitz
cell: 917 817-7396

BNNorman, 2019-02-20:
Windows 10, PyCharme CE2018.3.4. 
C:\WINDOWS\system32>java -version
java version "1.8.0_191"
Java(TM) SE Runtime Environment (build 1.8.0_191-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.191-b12, mixed mode)

Downloaded the mpfshell 1.1.1 zip, extracted it into the plugins folder. Went to Settings-plugins and enabled it. Went to settings-languages and frameworks enabled mpfshell support for ESP8266. Saved,  mpfshell initially appeared in tools menu greyed out. A few seconds later it was gone. Closed and restarted PyCharm - the tools menu had mpfshell in it, tried mpfshell REPL and got an IDE error  as follows. Same with mpfshell tools. Any suggestions?

java.lang.NoSuchMethodError: org.jetbrains.plugins.terminal.TerminalView.createNewSession(Lcom/intellij/openapi/project/Project;Lorg/jetbrains/plugins/terminal/AbstractTerminalRunner;)V
	at com.jetbrains.mpfshell.actions.MpfshellCommandAction.actionPerformed(MpfshellCommandAction.kt:45)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:258)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:275)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.lambda$actionPerformed$0(ActionMenuItem.java:287)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:283)
	at com.intellij.openapi.wm.impl.IdeFocusManagerImpl.runOnOwnContext(IdeFocusManagerImpl.java:106)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem$ActionTransmitter.actionPerformed(ActionMenuItem.java:277)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.lambda$fireActionPerformed$0(ActionMenuItem.java:111)
	at com.intellij.openapi.application.TransactionGuardImpl.runSyncTransaction(TransactionGuardImpl.java:88)
	at com.intellij.openapi.application.TransactionGuardImpl.lambda$submitTransaction$1(TransactionGuardImpl.java:111)
	at com.intellij.openapi.application.TransactionGuardImpl.submitTransaction(TransactionGuardImpl.java:120)
	at com.intellij.openapi.application.TransactionGuard.submitTransaction(TransactionGuard.java:122)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:111)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.doClick(BegMenuItemUI.java:522)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.access$300(BegMenuItemUI.java:35)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMouseInputHandler.mouseReleased(BegMenuItemUI.java:544)
	at java.awt.Component.processMouseEvent(Component.java:6548)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3325)
	at java.awt.Component.processEvent(Component.java:6313)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4903)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4725)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4725)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.awt.EventQueue.access$500(EventQueue.java:98)
	at java.awt.EventQueue$3.run(EventQueue.java:715)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.awt.EventQueue$4.run(EventQueue.java:735)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:719)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:664)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:363)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

JennaSys, 2019-03-16:
Same behavior  on Windows 7 with US keyboard

junhuanchen, 2019-04-03:
I'm back. I'm already PR

erpadmin, 2019-05-19:
was there a fix for this?

BNNorman, 2019-05-19:
Not yet AFAIK - I'm using uPyCraft now to upload micropython and use REPL - would rather it worked in PyCharm though.

junhuanchen, 2019-05-21:
In this problem, I use the [mpfshell-lite](https://github.com/junhuanchen/mpfshell-lite) as the underlying interfaces, like the [intellij-micropython-1.1.1.zip](https://github.com/vlasovskikh/intellij-micropython/files/3200353/intellij-micropython-1.1.1.zip).

you can separate test mpfshell-lite in Windows.

How come it's been so long without resolution. XD

hwiguna, 2019-05-30:
Andrey @vlasovskikh , fyi, I'm still running into this issue in version 1.0.11.
I'd be happy to beta test when you have a fix.  Thank you!

AnKun2, 2019-05-30:
> Andrey @vlasovskikh , fyi, I'm still running into this issue in version 1.0.11.
> I'd be happy to beta test when you have a fix. Thank you!

Have you tried this recommendation - https://github.com/vlasovskikh/intellij-micropython/issues/43#issuecomment-393134806 ?

Please try and post the results.

hwiguna, 2019-05-30:
> > Andrey @vlasovskikh , fyi, I'm still running into this issue in version 1.0.11.
> > I'd be happy to beta test when you have a fix. Thank you!
> 
> Have you tried this recommendation - [#43 (comment)](https://github.com/vlasovskikh/intellij-micropython/issues/43#issuecomment-393134806) ?
> 
> Please try and post the results.

I found my copy of microrepl.py at:
C:\Users\hwiguna\.PyCharmCE2019.1\config\plugins\intellij-micropython\scripts\microrepl.py

I replaced its content with the one you mentioned in comment #43.
Here is the result:
Device path COM19
Quit: Ctrl+] | Stop program: Ctrl+C | Reset: Ctrl+D
Type 'help()' (without the quotes) then press ENTER.

>>> import gc
>>> gc.?[K
Traceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax
>>> gc.mem_?[Kfree()
30896
>>>

The backspace actually works, but it echoes "?[k" in addition to doing the backspace.
FYI, I am running windows 10. Putty, rshell, uPyCraft do not have this issue (they have other problems :-))
I really like PyCharm + MicroPython plugin the best, except for this one annoying bug and I wish there is a keyboard shortcut to open MicroPython REPL.

Thanks for looking into this.

amuntobii, 2020-02-11:
Still exists, and is still the most annoying part of my workflow

jdjjm, 2020-06-07:
The issue is with miniterm.py in pyserial. 
[Here is a modified version of miniterm.py that I found that fixes the issue.](https://github.com/vgkits/vanguard-cli/blob/master/vgkits/miniterm.py)

I downloaded that file and added it to
_C:\Users\USERNAME\AppData\Roaming\JetBrains\PyCharm2020.1\plugins\intellij-micropython\scripts_ 
as miniterm.py

I then modified the microrepl.py in the same directory and replaced the line:
`from serial.tools.miniterm import Miniterm, key_description`
with:
 `from miniterm import Miniterm, key_description`

Now backspace works and so do the up and down arrow keys.

vlasovskikh, 2020-06-08:
@jdjjm Thanks! I believe the fix has been made in PR pyserial/pyserial#351 which has not been accepted yet. Unfortunately, there were no new releases of pyserial since 3.4 in 2017-07-24, although the project on GitHub seems to be at least partially alive.

I don't want to include the sources for pyserial and modify them, but I guess it's possible to monkey-patch the `Console` class in https://github.com/vlasovskikh/intellij-micropython/blob/master/scripts/microrepl.py.

jdjjm, 2020-06-08:
@vlasovskikh Yeah I that makes sense. Seeing as the pyserial PR was made all the way back in June 2018 maybe a monkey patch in the mean time is a good idea?

BNNorman, 2020-06-11:
I'd like to check this out. I'm using Pycharm on Windows 10

vlasovskikh, 2020-06-13:
@BNNorman Please follow the discussion in #105 for the versions to test and the steps to do it.

BNNorman, 2020-06-15:
Ok, I'm having problems installing this. 'I can't see the wood for the trees'  would be an appropriate thing to say. The documentation says I have to put microrepl.py into the PyCharm plugins folder - done that (inside its own sub-folder like all the other plugins) but not convinced pycharm (2020.1.2) is using it because I added something to the help text message (just to confirm) and I don't see that when I launch Tools-Micropython->micropython REPL.
There was no pre-existing microrepl.py as I had installed Pycharm 2020.1.2 recently from scratch after the last update clobbered pycharm. I have tried disabling the micropython plugin , close pycharm, open pycharm and re-enable the plugin.

I don't use micropython very often so I'd appreciate some 'for dummies' help with this because I like the PyCharm IDE but hate the non-working windows nav keys in the repl window.

Thanks for any help you can give.

jdjjm, 2020-06-15:
@BNNorman have you checked here: `C:\Users\USERNAME\AppData\Roaming\JetBrains\PyCharm2020.1\plugins\intellij-micropython\scripts`
That is where my files are stored. Not sure if it's different for standalone installs because I'm using the Jetbrains Toolbox.
Also the plugin calls the script each time you run the repl so you don't need to disable the plugin, you can just swap the python file and hit open repl. You are going to need to find the correct folder that already has the scripts in it. If the plugin is installed and you can't find the scripts folder with prexisting microrepl.py, microupload.py etc. then you probably have the wrong folder.

Let me know if you are still having issues finding the folder.

BNNorman, 2020-06-16:
Thank you ever so much for your help it is now working.
So I have two plugin folders:-
C:\Program Files\JetBrains\PyCharm Community Edition 2020.1.2\plugins
and
C:\Users\brian.000\AppData\Roaming\JetBrains\PyCharmCE2020.1\plugins\intellij-micropython\scripts

The microrepl stuff was in the latter - I never thought to look there. 

Having replaced the scripts with those from github the backspace,up,left and right arrow keys now function as expected (no weird escape characters)

Thanks very much for fixing that problem.

vlasovskikh, 2020-06-18:
The fix is in master. I'm going to release it soon.

closed 2020-06-18 by vlasovskikh