JetBrains/intellij-micropython #44: 'Official' ESP32 Support?

kaybee335, 2018-02-04:
@vlasovskikh - Can we declare the ESP32 as offocially supported now?  We now have a huge ESP32 success sample space of at least 2 users :-) 

I suggest that it be identified as a separate board/device type (`ESP32`) in the Frameworks menu as I suspect in the future there may be differences between the two that matter to the plugin (OTA/wireless support looks like it could be one of these areas from my early investigation).

In addition, I would like to suggest that the board/device type be passed as an optional parameter to the python scripts (possibly something like `--devtype ESP8266` on the command line.  We currently have `ESP8266`, `Micro:bit` and  `Pyboard`).  If not specified, it would default to None.

This would allow conditional execution of the couple of delay() statements in the scripts that are commented as being ESP8266 specific. 

I also have a feeling that this may help in the area of Micro:bit support although I have not explored that yet.  I plan on looking at the scripts in #40 today to see how/if they would fit into the existing scripting framework if you are not already doing so.  I dug out my Micro:bit so I should be able to check things out pretty readily.

I would be happy to take care of the python side of the new option as part of the docopt update.  It could be implemented initially with no 'behaviour' based on the option if you prefer that initially.  I am not comfortable enough with my weak, weak abilities to read kotlin to take care of the kotlin side of this so I would have to coordinate with you on the activity.

I know you are handling some of the board specifics in the kotlin code (i.e. defining vendor_id:port_id's to scan in order to detect serial ports) but it may be worth considering pushing all of this down to the python layer and creating an abstract devboard class that is then used to create concrete classes for the various board types.  The class interface would be optimised for plugin operations such as uploading a file to the device, removing files, REPL, etc.  

Just some thoughts for consideration, I recognise that I am much more comfortable in python so I could be trying to use my **hammer** in places it isn't wanted :-)  Please feel free to tell me to butt out on this one.

vlasovskikh, 2018-02-10:
@kaybee335 I still want to get my hands on a ESP32 board so I'll be able to investigate any possible issues before adding it as a device type. Meanwhile feel free to add more comments about the specifics of ESP32.

kaybee335, 2018-02-11:
@vlasovskikh - sure, understood.  I gave not hit any differences so far between ESP8266 and ESP32 that directly impact the plugin in its current state.  I do not believe this will remain true if a means of providing OTA/WiFi is added as an enhancement.  So far, I have found no common ground between the two in terms of supporting this (webREPL is 8266 only, etc.).  The other area of difference I have been bitten by is the removal the the _thread module from ESP8266 but this does not apply to the plugin.

If you are looking for recommendations on ESP32 boards, I am very happy with my Adafruit Feather ESP32 board.  It seems to have the right blend of built-on stuff and accessible IO for most of things I do, certainly at the prototyping stage.  The fact that I also have a couple of FeatherWings (OLED, GPS, relay) makes working with these quick and easy to check ideas out.

Another possibility would be the WiPy family of boards.  I have a WiPy 2.0 and it is nice board although you really want to get the Expansion (or other) board that provides the USB/UART in order to program it easily (the other option is an external USB/UART and some microswitches for the programming IO lines).
It has a specific micropython firmware image so it would likely require board specific typehinting work as well.

If you are looking for something cheaper, the Geekcreit boards on Banggood seem to get pretty solid reviews.  You would use the ESP32 daily build of micropython on these like the Adafruit board.  I have a couple on order but they are literally on the slow boat from China :-)  Note that if you are planning on using the deep sleep capabilities you want to make sure the ESP32 module has the updated device module on it otherwise some external circuitry is required.

https://www.banggood.com/ESP32-DevKit-ESP32-WiFiBluetooth-Development-Board-ESP-WROOM-32-Dual-Cores-Board-p-1138249.html

I also have a pyboard 1.1 on order so I should be able to help out on that side of things shortly as well.

gaitskell, 2018-02-11:
Thank you for pushing on with explicit ESP32 support. This micro controller is a really useful device  with strong/flexible communications and extended ADC/DAC support. (I'm using Adafruit ESP32 Feather boards.)

iascchen, 2018-03-02:
I am trying to use esp32 in pycharm. and wish this feature can be added. thx.

dmitrykudriavtsev, 2018-04-10:
atm we have two projects of micropython:
-official from micropython.org
-fork from micropython.org to https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/
last one is better by many things.
i`m using second firmware and after softreset by pycharm device not answering at all.

greenoid, 2018-04-25:
I would like to use my generic ESP32 board (Wemos Lolin32) with PyCharm, Micropython plugin and the fork from https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/
I'll give it a try and see where the problems are.

SensorsIot, 2018-05-01:
I also use the psRAM version on an ESP32. I get REPL, on COM64 but cannot run a program:
D:\Dropbox\MicroPython\Project1\venv\Scripts\python.exe C:\Users\andre\.PyCharmCE2018.1\config\plugins\intellij-micropython/scripts/microupload.py -C D:/Dropbox/MicroPython/Project1 -v COM64 D:/Dropbox/MicroPython/Project1/boot.py
Connecting to COM64
Traceback (most recent call last):
  File "C:\Users\andre\.PyCharmCE2018.1\config\plugins\intellij-micropython/scripts/microupload.py", line 138, in <module>
    main(sys.argv[1:])
  File "C:\Users\andre\.PyCharmCE2018.1\config\plugins\intellij-micropython/scripts/microupload.py", line 56, in main
    board = Pyboard(port)
  File "D:\Dropbox\MicroPython\Project1\venv\lib\site-packages\ampy\pyboard.py", line 147, in __init__
    raise PyboardError('failed to access ' + device)
ampy.pyboard.PyboardError: failed to access COM64

Ampy works at the command prompt. It seems that REPL blocks the Serial connection. 
What is my mistake?

MiLeG, 2018-08-19:
@SensorsIot I believe you did not close the REPL prompt before you uploaded. Serial programmers can't do their job while the port is busy. It's the same with Arduino and other serial communication.

tcinbis, 2018-10-03:
Would be great to see the ESP32 as a feature in the future.

Kortenbach, 2018-11-21:
> ampy.pyboard.PyboardError: failed to access COM64
> Ampy works at the command prompt. It seems that REPL blocks the Serial connection.
> What is my mistake?
You can look at issue "LED blink example #72" for this.
I found that running Pycharm in elevated mode sometimes does the trick as well...

junhuanchen, 2018-11-22:
support esp 8266 or 32 in mpfshell. you can try it. https://github.com/junhuanchen/intellij-mpfshell

IainColledge, 2019-03-12:
I'm using this plugin with the NodeMCU ESP-32S and also with the WIFI KIT 32, no problems apart from REPL on Windows 10 is useless as is the ESP8266. Tiny Term is the workaround.

x79133000, 2019-08-16:
Any news on the official ESP32 support yet?

vlasovskikh, 2019-08-19:
I still haven't got a ESP32 device to test it. Looking for contributors who can support ESP32 and fix possible bugs in this area.

jamesle-0812, 2019-11-27:
Hi everyone, I have a NodeMCU ESP-WROOM-32, shipped from China, and I wanna play around with MicroPython. So I've erased the flash and then flashed a firmware recommended on micropython.org for ESP32. From PyCharm, I've installed ampy and MicroPython plugins, set up to ESP8266 with a properly COM Port. I also run MicroPython REPL and then close the terminal before flash the demo code on [https://blog.jetbrains.com/pycharm/2018/01/micropython-plugin-for-pycharm/](url) to my ESP32 for blinking LED. The funny thing is it can connect to COM port, upload files 100% but then immediately do a soft reboot and the LED doesn't blink at all. 

> "C:\Users\James\PycharmProjects\ESP32\uPython 101\venv\Scripts\python.exe" C:\Users\James\.PyCharmCE2018.3\config\plugins\intellij-micropython/scripts/microupload.py -C "C:/Users/James/PycharmProjects/ESP32/uPython 101" -v COM9 "C:/Users/James/PycharmProjects/ESP32/uPython 101/main.py"
Connecting to COM9
Uploading files: 0% (0/1)
C:\Users\James\PycharmProjects\ESP32\uPython 101\main.py -> main.py
Uploading files: 100% (1/1)
Soft reboot

Is there any suggestion? I much appreciate your time and support!

vlasovskikh, 2020-06-05:
I kept postponing it since I didn't have any ESP32 devices to test it with, but it looks like my ESP8266 communication scripts should work for ESP32 as well, so as a starting point I decided to use them. ESP32 has a few changes in the device libraries and I'll eventually address them as well.

As a workaround you can select ESP8266 as your device type and most of the functions of the plugin should work for you.

Piranessi, 2021-03-23:
Can I safely choose device type ESP8266 instead of ESP32 in pycharm?

vlasovskikh, 2021-03-23:
@Piranessi Yes, you can. There will be some missing type hints, but it mostly works.

Piranessi, 2021-03-24:
> @Piranessi Yes, you can. There will be some missing type hints, but it mostly works.

Thanks!

MrNavaStar, 2022-12-20:
@vlasovskikh I have created type hints for the esp32. I have made hints for the [esp](https://micropython-glenn20.readthedocs.io/en/latest/library/esp.html) module (only functions actually supported by the esp32), and the [esp32](https://micropython-glenn20.readthedocs.io/en/latest/library/esp32.html) module. I am planning on also creating hints for the [espnow](https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html) module, however I do not believe these are yet in the main branch of micropython. Please correct me if I am wrong about that and I will create them right away.

What would be the best way to contribute them? Should I make a pr to a certain issue? or just a general pr?

For now my work is sitting on my [fork](https://github.com/MrNavaStar/intellij-micropython/tree/master/typehints/esp32). Any functions that are missing comments are due to there being no comments on the official docs.

Elizaveta239, 2022-12-21:
Hi @MrNavaStar! That sounds really cool! Could you please create one general PR for these changes? 

Then I'll add ESP32 as a separate device type and we will be able to include it into the next release. Thank you!

MrNavaStar, 2022-12-21:
@Elizaveta239 That sounds wonderful! I’ll make a PR later today.

elmot, 2024-04-04:
I am migrating from abandoned ampy to mpremote. When its done, I'll verify the PR

lukaskremla, 2024-12-22:
> Any news?

Technically yes: #330

bh1rio, 2024-12-22:
> > Any news?
> 
> Technically yes: #330

There is still only the esp8266 in the device list and no esp32.
Only support win？

lukaskremla, 2024-12-22:
> > > Any news?
> > 
> > 
> > Technically yes: #330
> 
> There is still only the esp8266 in the device list and no esp32. Only support win？

esp8266 is only for type hints (which are very unreliable anyway as of right now). You can use esp32 when having esp8266 selected.

For proper stub/type hint support manually look at this: https://micropython-stubs.readthedocs.io/en/main/

bh1rio, 2024-12-22:
Connecting... I/O error - Traceback (most recent call last): File " ", line 12, in   File " ", line 10, in fld File " ", line 10, in fld File " ", line 7, in fld OSError: [Errno 22] EINVAL

lukaskremla, 2024-12-22:
> Connecting... I/O error - Traceback (most recent call last): File " ", line 12, in File " ", line 10, in fld File " ", line 10, in fld File " ", line 7, in fld OSError: [Errno 22] EINVAL

Are you using the marketplace version of the plugin? The pre alpha? What were you doing to get this error? 

I really can't help you if you just give me a short error message.

bh1rio, 2024-12-22:
The pre alpha, installed form the zip file download in the  page

and where can I get the detail message?

lukaskremla, 2024-12-22:
> The pre alpha, installed form the zip file download in the page
> 
> and where can I get the detail message?

Make a post in the #330 discussion, things related to the pre-alphas go there.

If you could record a video of what you do to get this error then I can try to help you figure out what's wrong.