JetBrains/intellij-micropython #46: .git directory uploaded by project upload

dmopalmer, 2018-02-14:
Although the .idea directory is properly excluded when you flash a project directory, it does try to flash the contents of the .git directory, if it is at that directory level.

This is so even if .git is is listed in the excluded files.

vlasovskikh, 2018-02-17:
For me the plugin uploads `.git/` even though it's hidden from the project structure by default in both IntelliJ and PyCharm. I think the plugin should have ignored all hidden files in the project structure. I haven't checked yet whether this data is available via the public IntelliJ API.

@dmopalmer Since it's not visible in the project structure could you please tell how have you excluded it?

The obvious workaround is to put all your source code into a separate folder (e.g. `src/`), though it's a bad idea to customise your project structure just to make a tool happy.

dmopalmer, 2018-02-17:
In `preferences->project structure`, I tried adding it to `excluded files` at the bottom.  (Even though it would be an `excluded directory` if that existed.)

A more convenient option would be to have something like a `flashlist.txt` file that lists all the files that you should flash.

vlasovskikh, 2018-02-17:
@dmopalmer Yes, a way to provide the list of files to upload is another option. I guess I'll explore the idea of detecting hidden files first.

vlasovskikh, 2018-02-22:
This bug starts to hurt me too. I've found the API for ignored files listed in _"File | Settings | Editor | File Types | Ignore files and folders"_:

```kotlin
fun FileTypeManager.isFileIgnored(file: VirtualFile): Boolean
```

We'll have to move file filtering from microupload.py to the Kotlin side and pass the list of files to upload via stdin.

dmopalmer, 2018-02-22:
If you want to do the simplest thing, you can ignore files/directories that begin with `.` under Unix/Linux/Mac.  I think that  Windows has a 'hidden' attribute for files/directories that is accessible through some library.

TamiasSibiricus, 2018-04-08:
Small workaround:
anyone could add line of code into ./idea/YourProjectName.iml like this

```
<?xml version="1.0" encoding="UTF-8"?>
<module type="PYTHON_MODULE" version="4">
  ...
  <component name="NewModuleRootManager">
    <content url="file://$MODULE_DIR$">
      <excludeFolder url="file://$MODULE_DIR$/.idea" />
      <excludeFolder url="file://$MODULE_DIR$/.git" />
    </content>
  </component>
   ...
</module>
```

It works even after adding and saving new locations until removal from list in "File | Settings | Editor | File Types | Ignore files and folders"

vlasovskikh, 2018-05-30:
Related issue: #51.

vlasovskikh, 2018-05-30:
@max-kviatkouski Since the IntelliJ platform doesn't show ignored (as opposed to _excluded_) files in the project structure, it's hard for the user to mark their `.git/` folders and the like as excluded. The solution to explicitly mention them in the UI mentioned by you will work, but it would be better to do the right thing by default.

I suggest trying `FileTypeManager.ignoredFilesList: String` as a simple way to fix this issue without the need to move file filtering to the IntelliJ side. It will require recusively traversing the folder to be uploaded and matching file names against the patterns in this list.

If you do consider moving filtering from Python to IntelliJ, then `FileTypeManager.isFileIgnored(String)` might help.

cr0mbly, 2018-11-17:
Hi there interested in picking this up. @vlasovskikh are you happy for me to submit a pull request?

cr0mbly, 2018-11-20:
@vlasovskikh thoughts on applying the following (I don't know kotlin so let me know if there's anything that can be optimized)

Esp8266DeviceProvider.kt
```Kotlin
  override fun getRunCommandLineState(configuration: MicroPythonRunConfiguration,
                                      environment: ExecutionEnvironment): CommandLineState? {
    ...
    val excludes = excludeRoots
        .asSequence()
        .filter { VfsUtil.isAncestor(file, it, false) }
        .map { VfsUtilCore.getRelativePath(it, rootDir) }
        .filterNotNull()
        .map { listOf("-X", it) }
        .flatten()
        .toList()

    val excludedByEnvironment = FileTypeManager.getInstance().ignoredFilesList
            .split(';')
            .map { listOf("-X", it) }
            .flatten()
            .toList()

    val command = listOf(pythonPath, "${MicroPythonFacet.scriptsPath}/microupload.py", "-C", rootDir.path) +
        excludes + excludedByEnvironment + listOf("-v", devicePath, configuration.path)

```

works on my testing although I'm using an ESP8266 not sure how best to apply it to the other boards

dmopalmer, 2019-02-18:
(I don't know Kotlin and I don't know the build system for Intellij, so I am hacking the python in-place at 
`~/Library/Application Support/PyCharmCE2018.3/intellij-micropython/scripts/microupload.py` )

Adding code to `microupload.py::list_files()` at least gets rid of the .git directory.  However, the 'exclude-files' list doesn't seem to get passed to `microupload.py::main()`

```
def list_files(path: str, excluded: List[str]) -> Iterable[str]:
    """List relative file paths inside the given path."""
    # FIXME what if you want to include a hidden file?
    # FIXME exclude Windows Hidden files?
    excluded = {os.path.abspath(x) for x in excluded}
    for root, dirs, files in os.walk(path, topdown=True):
        abs_root = os.path.abspath(root)
        for d in list(dirs):
            if (os.path.join(abs_root, d) in excluded
                or os.path.basename(d).startswith(".")):
                dirs.remove(d)
        for f in files:
            if (os.path.join(abs_root, f) not in excluded
                and not os.path.basename(f).startswith(".")):
                yield os.path.relpath(os.path.join(root, f), path)
```

The workaround of putting the source files in a subdirectory such as `/src` is that they get flashed to the device as `src/*`, so you have to `import src.foo` to use them.

vlasovskikh, 2020-06-06:
@dmopalmer I'm sorry it took me so long to get back to this project :(

I've realised that your idea to take a simple approach and just ignore hidden files starting with '.' will work in most contexts. For visible directories the plugin already respects excluded folders. Yes, there are still *.pyc, *~, __pycache__, and other stuff PyCharm ignore via wildcard patterns from _"File | Settings | Editor | File Types | Ignore files and folders"_, but they should be rare in the context of MicroPython projects.

vlasovskikh, 2020-06-06:
Done. I'll publish an updated version of the plugin soon.

closed 2020-06-06 by vlasovskikh