JetBrains/intellij-micropython #48: REPL, line edit, history, releasing COM port

otalado, 2018-02-28:
Not really issue but more of usability issues: 
- I really wish that it would be possible to fix typing errors when in REPL
- I wish to be able to recall lines from history when in REPL
- I often get stuck with COM port not being closed/released. Very annoying. I wish to have an obvious way to see status of COM port and a way to release it (tools button/context menu ?).

Other than that, I find micropython plugin **fantastic tool.** Congratulations!

otalado, 2018-03-01:
Hi, I have to retract my comments above. I found an explanation for my situation in issue #43. It turns out it is happening on W10 platform. I installed PyCharm on Raspberry PI /Raspbian (Debian) 9 - everything works OK!
I hope this will be solved for W10 sooner or later, because I can use the RPI installation, but it is slooow...

vlasovskikh, 2018-03-01:
@otalado Thanks for your comments! Closing this one as a duplicate of #43.

closed 2018-03-01 by vlasovskikh