JetBrains/intellij-micropython #49: REPL on Raspbian(Debian) Stretch

otalado, 2018-03-02:
I installed PyCharm Community 2017.3.3 and micropython plugin  Raspberry Pi Raspbian (Debian) Stretch. It was working OK, but it tended to freeze a lot. PyCharm itself suggested that the installed version of Java (1.8.0_65 build17) may be causing the problems. So I downloaded and installed the latest JDK (1.8.0_162 build 12).
Ever since I am unable to start the REPL. It just shows gray terminal window - and that's it.
I am not sure if version of Java is to blame. I remember that PyCharm suggested to install lower than the one I installed, but unfortunately I did not follow it.
So, can you, please, tell me which Java version is recommended/required? And, just in case that Java version is not the problem - what do you suggest I should check?

vlasovskikh, 2018-05-30:
@otalado Sorry for my late reply. Usually the best version of Java is the one bundled with PyCharm. We recommend switching to another Java version only if there are specific known issues that are caused by a particular version of JVM. Try PyCharm Community 2018.1.3 with its bundled Java. Did it help?

vlasovskikh, 2018-07-24:
Closing this one since there hasn't been any activity in this issue for a while.

closed 2018-07-24 by vlasovskikh