JetBrains/intellij-micropython #50: Strange symbols after hard reset in REPL

TamiasSibiricus, 2018-03-16:
Hi, nice work for esp8266 support
But there is some problem. Sometimes after device hard reset REPL show strange symbols(take a look at screenshot). It is result of print() function into stdlog(please correct me if i wrong). Sometimes result looks readable. I do not use any any language for debug messages except English.
Bad symbols shown randomly.
OS: Linux Ubuntu 16.04
pyCharm: 2017.3.4 (Community Edition)
intellij-micropython: 1.0.6
Board: NodeMCU v3

[image]

vlasovskikh, 2018-04-06:
@TamiasSibiricus I've been experiencing this issue as well. I guess ESP8266 devices sometimes write some extra bits to their serial ports before launching a proper terminal session. I haven't looked into the details, but I've seen the same problem with regular command line sessions with ESP8266 devices as well.

TamiasSibiricus, 2018-04-08:
Additional info:
it is possible to get back to normal fonts if you close REPL and try open new. In other words complete REPL restart could fix problem. Sure until it happens again.

vlasovskikh, 2018-05-30:
I keep observing these strange symbols outside of PyCharm in a regular terminal session. It's likely an issue with the serial port output of devices, not in terminals we use to connect to them.

Walkline80, 2018-10-20:
this issue still exist in pycharm with this plugin for esp8266......

vlasovskikh, 2018-11-12:
@Walkline80 Don't you experience it in the system terminal outside PyCharm? I see it from time to time, so I believe the problem isn't related to the MicroPython plugin for PyCharm.

rftestman1, 2018-11-18:
I am having this same issue on win10/Pycharm 2018.1.  Never seen a problem when communicating serially using Putty.

Kortenbach, 2018-12-15:
I'm having the same issue. When I close the PyCharm REPL and open PUTTY everything is working fine. I think it has to doo with the plugin. Maybe stopbits mismatch(?)