JetBrains/intellij-micropython #51: Flashing of project does not ignore excluded folders

roamingthings, 2018-03-17:
Hi,

I've setup the plugin to flash all files of my project excluding the .idea folder as described. However the plugin still uploads the .idea folder to the pyboard.

Any thoughts?

RoundBobin, 2018-03-20:
I'd like to provide a way around if you like.

cd ~/Library/Application\ Support/PyCharmCE2017.3/intellij-micropython/scripts/
edit microupload.py file and add something like this

excluded_files = opts['--exclude']
excluded_files.extend(['.git','.DS_Store'])
...
    for x in list_files(root, excluded_files)]

vlasovskikh, 2018-05-30:
@RoundBobin A simple solution like hard-coded .git/, .DS_Store/ and other folder could work. For an alternative fix idea that takes into account the actual file patterns ignored by IntelliJ see my comments in #46.

vlasovskikh, 2018-05-30:
Closing this issue as a duplicate of #46. Please vote for the new issue using :+1:.

closed 2018-05-30 by vlasovskikh