JetBrains/intellij-micropython #52: Support for Pycom boards

cvrtky, 2018-03-31:
Hello Andrey
the guys at pycom.io have bailed on their promisses on creating a PyCharm plugin for their MicroPython boards. Would you consider adding support for these boards ?

Sincerely
Tim

vlasovskikh, 2018-04-01:
@cvrtky Yes, it's a good idea to include support for Pycom devices to the existing intellij-micropython plugin. I could help with the plugin development questions and I'd accept pull requests that add new devices / new features.

pvangeel, 2018-12-06:
I wanted to have a look as I own a wipy as well. I was just wondering: The typing hints, are they manually created? Or is there a way to extract them from the micropython port of pycom?

vlasovskikh, 2018-12-06:
@pvangeel I've created them manually based on the API docs for modules and refining this information with what I learned by experimenting the the API interactively in a REPL session. Interactive experiments mostly helped to come up with correct type hints for parameters and return values.

pvangeel, 2018-12-06:
Cool thx. I'll check if I can find some spare time. Would be really nice to have this working instead of using an Atom plugin!

GonzRon, 2019-02-28:
+1 on this guys, i just got a wipy 3.0, anyway I can make this work manually?  I am using pycom vsCode plugin for now but would love to switch to pyCharm

thomasdf, 2020-08-26:
Has there been any progress on this? I too really prefer Pycharm over Atom and VSCode.