JetBrains/intellij-micropython #53: Close REPL on flashing

TamiasSibiricus, 2018-04-08:
I have used your plugin for some days and found it very nice. It is better than using ESPlorer for me.
But there one thing that annoying too much:
When i open repl for debug purposes and then want to reflash device after some time, i need alway close it and flash device. Ok. But in 40-50% of my actions like described cycle i completely forget to close REPL before flashing process.
So there is possible solutions:
- close opened REPL(usuall only one REPL opened for 1 device), flash device, and open REPL for device again.
- use ESPlorer way(too buggy as for me) drop and write files line by line via REPL if it opened.

Sure it is only feature proposal and should discussed... Thanks

dkayser, 2018-04-09:
The same issue happens to me with an ESP32. I have to manually press the reboot button to make it work again after each upload, or if a REPL was active in the background. It seems that entering raw REPL mode does not work reliably. It worked really well with the generic 8266 I used before. This is on Debian testing. Tried different USB cables, no change.

I looked into contributing better ESP32 support, if I could get some pointers of where to start it looks feasible.

TamiasSibiricus, 2018-04-09:
@dkayser I think reboot issue is very depend on board that in use. Mostly it is for chip used as bridge between USB and RS232 port. I have few boards with different bridges and can confirm that soft reboot not always work on all boards. My last development board 
[image]
have pin for reseting and bridge that comes with board send soft reset signal with good result. :)
Hard reseting for this case needed only if i cleanup all files from device and write completely new.
Right now plugin successfully soft reboot my card as well.

dkayser, 2018-04-09:
Interesting thanks for sharing your findings. My UART - USB driver is:

    $ lsusb
    Bus 007 Device 002: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR 
    mySmartUSB light

    $ dmesg
    usb 7-1: new full-speed USB device number 3 using uhci_hcd
    usb 7-1: New USB device found, idVendor=10c4, idProduct=ea60
    usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    usb 7-1: Product: CP2102 USB to UART Bridge Controller
    usb 7-1: Manufacturer: Silicon Labs
    usb 7-1: SerialNumber: 0001
    cp210x 7-1:1.0: cp210x converter detected
    usb 7-1: cp210x converter now attached to ttyUSB0

and 

    $ lsmod | grep cp210
    cp210x                 32768  0
    usbserial              53248  1 cp210x
    usbcore               290816  12 cp210x,usbhid,rt2800usb,snd_usb_audio,usb_storage,rt2x00usb, 
    ehci_hcd,uhci_hcd,usbserial,snd_usbmidi_lib,uas,ehci_pci

Well it is probably a bad knock off... Maybe I try returning the board for a refund and try a different manufacturer.

TamiasSibiricus, 2018-04-10:
@dkayser Sure it is offtopic, but... I really doubt that you get refund for working board. Main problem with chineese boards is too much fakes or replicas. :) I see your board use CP2102 chip as bridge and it is better than ch430/431. But your card could be just replica of original NodeMCU or any another board with esp32 chip. So it could just do not have fine soldered control circuit for soft reseting. for TTGO  experimental board(it is not advert :) ) i was really surprised for working soft reset. So just buy original boards and all things should work.

juliandroid, 2019-08-23:
Is there a workaround for this? I'm using the console all the time for debugging and I have always to close it before reflashing it and open the console again.

vlasovskikh, 2020-09-21:
It will be easier to implement if #101 is added.

silmaril42, 2020-11-28:
I love the idea that flashing should close REPL or reuse it's windows / process.

Additionally I would add the wish to (optionally) automatically open REPL after flashing. \
This way we would be able to see the complete output of the boot process without missing the first seconds that we need to open REPL manually. 

Should this be a separate issue?

vinraspa, 2022-08-15:
Anything new about this?
Before working with pyCharm and µpython, I used to program in C++ with atom and platformio. As soon as you upload a new sketch, it automatically closes the serial monitor and reopen it just after upload is done.

elmot, 2024-07-24:
Fixed, Duplicate #199

closed 2024-07-24 by elmot