JetBrains/intellij-micropython #57: No ADC class in machine.pyi

andreas-harmuth, 2018-06-04:


vlasovskikh, 2018-06-05:
@andreas-harmuth What MicroPython port do you use? Could you please point to the documentation page that describes the class you're referrring to? My quick search gave me `pyb.ADC`, not `machine.ADC`.

andreas-harmuth, 2018-06-10:
Hi @vlasovskikh, I'm not completely sure what you mean by port. I'm using MicroPython for ESP8266 and ESP32, and found the documentation for machine.ADC  [here](http://docs.micropython.org/en/v1.9.3/esp8266/library/machine.ADC.html)

vlasovskikh, 2018-06-14:
@andreas-harmuth Thanks for the link! For some reason `machine.ADC` is not mentioned at [the page](http://docs.micropython.org/en/v1.9.3/esp8266/library/machine.html) about the `machine` module for ESP8266.

By ports I meant the CPU type of your board. It's the terminology used by MicroPython, see their [project structure](https://github.com/micropython/micropython/tree/master/ports).

JennaSys, 2019-03-15:
The latest version (1.10) [docs page](http://docs.micropython.org/en/latest/library/machine.html) now lists ADC.   PyCharm inspection is also missing machine.PWM however, which is still not listed in the docs for machine.  (I'm using 2018.3.5, ESP8266, Python 3.7.2)

hlovatt, 2021-08-08:
The current version of Micropython uses `machine.pyi` from https://github.com/hlovatt/PyBoardTypeshed which includes both ADC and PWM, if you upgrade the plugin does the problem go away?

hlovatt, 2021-10-30:
@fablableo Thanks for finding this mistake. I fixed up this error and a couple of others that I found as a consequence of improving the error checking to find the original mistake you reported!

You can install the fixed typesheds manually, see Manually install in IDE in README on https://github.com/hlovatt/PyBoardTypeshed.

vlasovskikh, 2021-12-01:
Should be fixed by #176.

closed 2021-12-01 by vlasovskikh