JetBrains/intellij-micropython #60: Pin class is not included completely

Streyder, 2018-06-21:
Hey,

the Pin class is sadly lacking some functionality being discovered. For example Pin.board.X, Pin.high() and Pin.low() are missing.

Cheers
Kevin

vlasovskikh, 2018-07-24:
@Streyder Thanks for reporting it. A pull request with missing type hints is welcome.

Streyder, 2018-08-09:
I implemented the changes on my end. Sadly I am not allowed to create a new branch to create a pull request. Actually I don't know how I would make a proper pull request. Maybe you can help out.

vlasovskikh, 2018-08-09:
@Streyder We use the standard GitHub workflow for pull requests to this repository, I'm sure there are good tutorials about pull requests on GitHub. The basic idea is that you close this repository to your account, so it becomes github.com/Streyder/intellij-micropython, then you create your new branch there and then GitHub will show you the message about creating a pull request to the original repository on the page of your clone.

Streyder, 2018-08-10:
Created pullrequest [63](https://github.com/vlasovskikh/intellij-micropython/pull/63) for this issue

vlasovskikh, 2020-06-06:
Closing it as a duplicate of #16.

closed 2020-06-06 by vlasovskikh