JetBrains/intellij-micropython #62: How to flash all project files

kylinpoet, 2018-07-22:
I had done follow the instruction：
> Select the top level folder of the project in the path browser of "Run/Debug Configurations".

BUT it shows：

> Python files must end in ".py".

vlasovskikh, 2018-07-24:
Make sure you've selected a "Flash ...", not a regular "Python" run configuration.

kylinpoet, 2018-07-25:
> Make sure you've selected a "Flash ...", not a regular "Python" run configuration.

Sorry for my poor  understanding、、、
I use the pycharm for windows:
[image]
There's something different。。。
Could you show more tips?
囧囧囧

vlasovskikh, 2018-07-27:
Everything looks correct so far. Please edit one of your t1.py or r1.py run configurations and try to specify the path to your project directory instead of a specific *.py file. Please attach a screenshot if you see any errors you don't expect.

kylinpoet, 2018-07-28:
[image]

What I had done are these:
[image]
What's the next now??
Sorry again for disturbing you...

vlasovskikh, 2018-08-01:
@kylinpoet I got it now. Uploading multiple file to Micro:bit devices is not supported yet. It works only for ESP8266 and Pyboard. Sorry about that. A pull request that adds support for it is welcome. It will require changing the upload method from `uflash` so some other packages that actually allows transferring multiple files.

adam-binks, 2018-11-02:
@vlasovskikh You mentioned here that the upload method would need to be changed from `uflash` to some other packages that allow transferring multiple files, do you happen to know of any that would achieve this? 

I'm embarking on a big project using a micro:bit which I really don't want to develop all in a single file! I'm unsure how to flash multiple files though.

Cheers!

vlasovskikh, 2018-12-06:
@adam-binks Sorry, I'm not aware of any. I haven't done any research, so it may be possible that a tool for uploading multiple files to Micro:bit devices exists. AFAIK `uflash` allows to upload only a single Python file built into the binary of the MicroPython interpreter itself.

carlosperate, 2019-03-03:
https://github.com/ntoll/microfs/ can be used to add multiple files to the micro:bit MicroPython filesystem.

marcelloromani, 2023-03-05:
> https://github.com/ntoll/microfs/ can be used to add multiple files to the micro:bit MicroPython filesystem.

In a sample project I was able to successfully use `ufs` to upload a multi-file Pyhon project to the microbit.
One detail I found is that becase a file is executed straight away, any dependencies must be uploaded first.
Example from the Makefile of my test project:

```make
flashit:
	ufs put src/config.py
	sleep 5
	ufs put src/main.py
```

`main.py` contains `from config import MAX_COUNT`, therefore `config.py` has to be on the microbit already when `main.py` uploaded.