JetBrains/intellij-micropython #64: about win10 repl can used of my mpfshell.

junhuanchen, 2018-09-28:
https://github.com/junhuanchen/mpfshell

vlasovskikh, 2018-12-06:
@junhuanchen What are the benefits of using mpfshell instead of adafruit-ampy and pyserial? Could issues on Windows 10 and wireless uploading be fixed by smaller changes without switching to another library for communicating with devices? adafruit-ampy and pyserial are widely used libraries. I have doubts about stability of mpfshell and the community around it.

junhuanchen, 2018-12-07:
HI, good morning, here I am.

All right, let me explain. A couple of questions.

First, both adafruit - ampy mpfshell, the bottom is in [pyboard. Py] (https://github.com/micropython/micropython/blob/master/tools/pyboard.py), the dependence of pyserial.

Second, win10's repl control of messy code problem in pyserial console, it has nothing to do with adafruise-ampy or mpfshell.

Third, mpfshell differs from adafruit-ampy in that it supports more Linux commands, such as put/get/cat, including freely combined commands.

Finally, the mpfshell part I fixed is not only the Windows console problem of pyserial, but also the stable connection problem of pyboard.py (when connecting different devices), which is at least better than adafrutis-ampy, because the official connection board sub-process has not been fixed, you can have a try.

Because I understand the underlying communication mechanism of the board, I can fix all kinds of abnormal situations.

vlasovskikh, 2020-06-06:
Sorry @junhuanchen, I don't consider switching to mpfshell-lite, as I mentioned in #88.

closed 2020-06-06 by vlasovskikh