JetBrains/intellij-micropython #65: machine.Timer don't have init method?

Walkline80, 2018-10-14:
I'm using micropython for esp8266, fireware v1.9.4, I wonder that the machine.Timer only have two method: __init__ and deinit, it doesn't matching the micropython docs, and I can't using init() method to initializing a new Timer, what's the matter? please, help~

junhuanchen, 2018-10-15:
Well, it's only code highlighting tip not just, in fact within and can use the machine Timer init, you use directly.

Or 

Need to check your code?

Walkline80, 2018-10-15:
> Well, it's only code highlighting tip not just, in fact within and can use the machine Timer init, you use directly.
> 
> Or
> 
> Need to check your code?

Thanks for your reply, I'm according to official doc to init the Timer, my code is

`wifi_status_timer.init(peroid=1000, mode=Timer.PERIODIC, callback=check_wifi_status)`

and the throwed exception is

`TypeError: extra keyword arguments given`

and here is the official doc

`tim.init(period=2000, mode=Timer.PERIODIC, callback=lambda t:print(2))`

please help :)

closed 2018-10-20 by Walkline80