JetBrains/intellij-micropython #66: Possible analog_period_microseconds() issue.

md00bm, 2018-10-27:
Hi,
Can't seem to set analog_period_microseconds() with value less than 1000. The error message says: "ValueError invalid period". Can anyone confirm this?

pin0.set_analog_period_microseconds(500) # Cannot set below 1000?

junhuanchen, 2018-10-28:
Hi

This question should be asked by micropyhton at microbit.

md00bm, 2018-10-28:
Hi,
The Micro:bit team does not find anything wrong with the code under Mu, only in Pycharm with this MicroPython plugin.

The reply from the Micro:bit team:
Thanks for contacting micro:bit. I can't seem to recreate the error using Mu with micropython v1.0

carlosperate, 2018-10-30:
The uFlash dependency is likely at version 1.1, which would be an old version of MicroPython. https://github.com/vlasovskikh/intellij-micropython/blob/818f868da975f5c51b168db183fa171f9e31fe42/src/main/kotlin/com/jetbrains/micropython/devices/MicroBitDeviceProvider.kt#L46

If the issue cannot be replicated in Mu v1.0 (containing the latest micro:bit MicroPython v1.0.0) then issue was likely a problem already fixed in micro:bit MicroPython.

md00bm, 2018-10-30:
Hi Carlos,
I have searched for uFlash and update the version. But PyCharm seems to install the old version (1.0.8) in the project:

[image]

How can we solve this problem? Thanks for the input.

vlasovskikh, 2018-10-31:
If the issue is only in the version of uflash used by the plugin, then a pull request is welcome. @md00bm has mentioned the place in the code where the change is required.

md00bm, 2018-11-02:
Dear Andrey,
I have tried to contribute by creating the pull request for newer version of uflash as per suggestion. Totally not sure what I am doing, just edited the line metioned by @carlosperate:
from:
return manager.parseRequirements("""|uflash>=1.0.8,<1.1
                                        |pyserial>=3.3,<4.0""".trimMargin())
to:
return manager.parseRequirements("""|uflash>=1.2.3,<=1.2.3
                                        |pyserial>=3.3,<4.0""".trimMargin())

If there is any mistake please just correct it, anyone. Thanks.

Right now whenever I click on the "install requirements":
[image]

it will automatically install uflash 1.0.8 section:
[image]

Is there anything I can do to change that to uFlash 1.2.3?

Another error I encounter when creating micro:bit projects is "No devices detected.", but micro:bit is shown in File Explorer:
[image]

But the error does not affect file flashing though.

As for the original issue of this thread, analog_period_microseconds(), I think it is a separate issue from uFlash versioning. But can only know after the plugin installs the uFlash section showing 1.2.3, and test the code again.

vlasovskikh, 2018-12-06:
@md00bm Sorry for my late response. Please try to update `uflash` for your virtualenv manally to check if the most recent version has fixed the problem indeed.

You can do it in PyCharm via the list of packages in "File | Settings | Project | Project Interpreter". Select the line with `uflash` and invoke the "Upgrade" action. Alternatively, you can activate your virtualenv in the system console and run `pip install -U uflash` there. Just make sure you've activated the correct virtualenv via its </path/to/your/virtualenv>/bin/activate script.

Please tell if upgrading to uflash 1.2.3 has helped.

vlasovskikh, 2018-12-06:
@md00bm Regarding the second problem with the device detection process (by the way, I would really prefer to have separate issues for unrelated problems): device detection is based on USB vendor ID / product ID values for a serial controller of your MicroPython device. For Micro:bit we have now only a single pair of vendor ID / product ID, see https://github.com/vlasovskikh/intellij-micropython/blob/master/src/main/kotlin/com/jetbrains/micropython/devices/MicroBitDeviceProvider.kt#L42.

You can determine your vendor ID / product ID by exploring the results of `comports() from `serial.tools.list_ports`. See [this helper script](https://github.com/vlasovskikh/intellij-micropython/blob/master/scripts/findusb.py) for an idea of how to explore your connected USB devices.

closed 2020-06-05 by vlasovskikh