JetBrains/intellij-micropython #69: No connection to esp8266

Kepete, 2018-11-17:
I have got an esp8266 (NodeMCU v2) board and have flashed it successfully with esptool.
Also I can get the REPL connection to it using screen on mac. 

PyCharm version 2018.2.5 community edition and micropython plugin 1.0.9 installed. 
Under PyCharm -> Preferences -> Languages and Frameworks -> Micropython I have enabled the support, added device type and device path. The auto detect doesn't work. 
Should I use `/dev/cu.SLAB_USBtoUART` or `/dev/tty.SLAB_USBtoUART` as path? 
Excluded the PyCharm .idea directory.

Seems that there is no connection. boot.py, main.py an other don't appear in the IDE. 
Also in Tools there is no MicroPython REPL option. 
[image]

RoundBobin, 2018-11-21:
Had the similar issue.

Try to upgrade corresponding modules

pip3 uninstall serial
pip3 install pyserial --upgrade
pip3 install click --upgrade
pip3 install esptool --upgrade
pip3 install adafruit-ampy --upgrade

Kepete, 2018-11-22:
Thanks for the suggestion. Tried all of it and updated everything but no luck :/

RoundBobin, 2018-11-22:
Also please note that PyCharm 2018.3 does not work with plugin right now.
I tried it yesterday.
ver 2018.2 works fine.

Kepete, 2018-11-22:
Actually micropython plugin says:
```
Change Notes
1.0.9:
Compatibility with 2018.2-2018.3
```
But to be sure I tried 2018.2 with micropython plugin version 1.0.9 and didn't work.

vlasovskikh, 2018-11-22:
I've just published 1.0.10 that should be compatible with 2018.3. There was an issue with backwards incompatible API changes in 2018.3, so I had to limit the plugin to the latest version. For 2018.2 there is still branch 182 in the GitHub repo that is compatible with it.

vlasovskikh, 2018-12-06:
Closing this issue as resolved in version 1.0.10. Feel free to post a comment here if there are any issues left for 1.0.10 or newer.

closed 2018-12-06 by vlasovskikh