JetBrains/intellij-micropython #71: Setting device path (COM4) on Windows machine

Kortenbach, 2018-11-19:
How do I set the device path to my NodeMCU board on a windows machine (COM4) with file-->Settings --> MicroPython?

rftestman1, 2018-11-20:
Hi,

Apparently you would just put 'COM4' in the Device Path field as shown here:

[image]

But if you noticed my post from yesterday I am having some comms issue.  The REPL is working so the comms is partially working.  Let me know if you get it working.

Chris

Kortenbach, 2018-11-20:
Thank you for your reply! Nice to get a confirmation that putting com4 should do the trick. I posted a second issue in which I'm asking for a hello blinky example cuz I cannot get it to work either... Anyone wants to share his to get a blinking LED working under Win10?

Kortenbach, 2018-11-21:
This issue is resolved. Please look at issue "LED blink example #72"

closed 2018-11-21 by Kortenbach