JetBrains/intellij-micropython #72: LED blink example

Kortenbach, 2018-11-19:
I found the learning curve to be quite steep if your not familiar with PyCharm. Could you provide an LED blink example, just to get up & running?

rftestman1, 2018-11-20:
Have you seen this?:

https://blog.jetbrains.com/pycharm/2018/01/micropython-plugin-for-pycharm/

What board are you using?

rftestman1, 2018-11-20:
Right click the project in the project window:

[image]

Then select File and select the python file.

Pycharm is easy to use but takes some getting used to.  I would try some easier examples and come back to this one.  

I was hoping some of the experts would chime in on the comms problems but so far none have.

Kortenbach, 2018-11-20:
Thanks! And how do I transfer the file? (Or is this part of the comms problem you mentioned?)

rftestman1, 2018-11-20:
Yes, if working properly you just click the play button and the file is flashed to the chip.

[image]

Kortenbach, 2018-11-20:
Thanks, I will look at this tomorrow, see if I get it to work! I'll tell if it worked.

Kortenbach, 2018-11-21:
Ok, I'm starting to get the hang of it but it doesn't work...yet...
This is what I got:
C:\Users\user\PycharmProjects\TestPyCharm\venv\Scripts\python.exe C:\Users\user\.PyCharmCE2018.2\config\plugins\intellij-micropython/scripts/microupload.py -C C:/Users/user/PycharmProjects/TestPyCharm -v COM4 C:/Users/user/PycharmProjects/TestPyCharm/Blinky.py
Connecting to COM4
Traceback (most recent call last):
  File "C:\Users\user\.PyCharmCE2018.2\config\plugins\intellij-micropython/scripts/microupload.py", line 138, in <module>
    main(sys.argv[1:])
  File "C:\Users\user\.PyCharmCE2018.2\config\plugins\intellij-micropython/scripts/microupload.py", line 56, in main
    board = Pyboard(port)
  File "C:\Users\user\PycharmProjects\TestPyCharm\venv\lib\site-packages\ampy\pyboard.py", line 147, in __init__
    raise PyboardError('failed to access ' + device)
ampy.pyboard.PyboardError: failed to access COM4

rftestman1, 2018-11-21:
So you are getting the same errors I was.  Just got it working.

You need to make sure the MicroPython REPL session is closed in order to release the serial port.

Output should look like this:

[image]

Kortenbach, 2018-11-21:
I'm not getting further than:
C:\Users\user\PycharmProjects\TestPyCharm\venv\Scripts\python.exe C:\Users\user\.PyCharmCE2018.2\config\plugins\intellij-micropython/scripts/microupload.py -C C:/Users/user/PycharmProjects/TestPyCharm -v COM4 C:/Users/user/PycharmProjects/TestPyCharm/Blinky.py
Connecting to COM4
Uploading files: 0% (0/1)
C:\Users\user\PycharmProjects\TestPyCharm\Blinky.py -> Blinky.py
And then it stops...
I also found that the COM port does't get released properly after use, so the next try it will raise the exception...

Kortenbach, 2018-11-21:
Ok, now it works. Turns out my NodeMCU didn't like something I did and decided to stop working at some point. This made the upload proces fail and the COM port was left open because of that. I had to kill either Pycharm or the Python process to get the COM port to close again. Once I restarted my NodeMCU all was well. 
Thanks for the help!

closed 2018-11-21 by Kortenbach

rftestman1, 2018-11-21:
Glad you got it working.   Was good to know someone else was having similar issues.  And I was not alone.

Kortenbach, 2018-11-22:
Same here! Thanks again!

ShivaniButle, 2020-07-17:
> So you are getting the same errors I was. Just got it working.
> 
> You need to make sure the MicroPython REPL session is closed in order to release the serial port.
> 
> Output should look like this:
> 
> [image]

My output is looking like this same but led is not blinking