JetBrains/intellij-micropython #73: After update the neopixel is missing

Monsoon117, 2018-11-26:
I can't find the neopixel library on micro:bit after update to 2018.3.

Monsoon117, 2018-11-27:
The code can run. But the IDLE

vlasovskikh, 2018-12-06:
It's not reproducible for me with a project in PyCharm 2018.3, intellij-micropython 1.0.10, and a Python 3.7 virtualenv.

What are the exact errors for the two lines marked in yellow on your screenshot?

JennaSys, 2019-03-15:
I'm using PyCharm 2018.3.5, ESP8266, MicroPython plug-in v1.0.11, Python 3.7.2 on Windows 7 and inspection is reporting "No module named neopixel".

hlovatt, 2021-10-17:
Isn't the problem that you don't have a pyi (type stub) for NeoPixel. See https://github.com/hlovatt/PyBoardTypeshed for the required pyi file. You can manually install, see README at referenced GitHub address. Hopefully will become part of the plugin; have issued a pull request.

vlasovskikh, 2021-12-01:
Actually we already bundle `neopixel` with the MicroPython plugin. Please make sure you have selected Micro:bit as your device type in the MicroPython settings in PyCharm.

closed 2021-12-01 by vlasovskikh