JetBrains/intellij-micropython #74: Doesn't match documentation

hansonap, 2018-11-27:
Maybe I'm missing something, but this plugin seems to not recognize many of micropython's modules. For example `from machine import Pin, ADC, TouchPad` recognizes Pin but not ADC or TouchPad. Other modules I've come across that are not recognized:
* onewire
* ds18x20
* ubinascii

I'd be happy to help remedy this situation - although I've never worked with PyCharm plugins so will need a little pointing in the right direction.

junhuanchen, 2018-12-07:
in here https://github.com/vlasovskikh/intellij-micropython/tree/master/typehints.

Maybe you need to add it yourself.

Try to write pyi based on other libraries.XD

Josverl, 2023-03-20:
[Micropython stub packages on PyPI](https://pypi.org/search/?q=-stubs&o=&c=Programming+Language+%3A%3A+Python+%3A%3A+Implementation+%3A%3A+MicroPython)