JetBrains/intellij-micropython #77: File uploading is stuck

oriyair1, 2018-12-13:
I followed every step in the README file, though when I press run to upload the python files I get:
"Connecting to COM4
Uploading files: 0% (0/1)
C:\Projects\MicroPython\ESP32\main.py -> main.py"
This just stays like this forever.
Please tell me what to do.

oriyair1, 2018-12-13:
OK, thanks a lot! I will try to do that and be back with results.
I have one more question - what is the python version of micropython?

oriyair1, 2018-12-13:
@vlasovskikh 
I debuuged the microupload.py script and I noticed the this line is blocking the script - "files.put(remote_path, fd.read())"
Can you tell why?

jeffkarney, 2018-12-16:
Is your REPL terminal open when you are trying to upload? I noticed that if it is open, uploading will get stuck like you describe.

oriyair1, 2018-12-16:
Unfortunately no

L4grange, 2019-02-04:
I've also faced this issue, with one ESP32 board, which other tools were able to connect and flash files to. However, trying with a different ESP32 board, I was able to get the tool to flash the files sucessfully. I'll post here if I do more digging on the "bad board"

Simaho99, 2019-07-11:
did you find the solution on this issue

L4grange, 2019-07-15:
For me, I ended up using a different board and the upload is working fine ever since... Did not have the time to dig deeper into the issue.

Simaho99, 2019-07-15:
Great, but I have a solution for that, I have d one this to avoid that every time and it alwasys work.
1. When you are uploading make sure the running terminals, like "REPL terminal” or “Python terminal” are all closed. These terminals are found on the bottom of the IDE..
2. If for some reason it doesn’t upload, I found that if you flash it again with esptool.py on the terminal, it works fine. (You can create a script to do that, It will be easy)

vlasovskikh, 2020-06-06:
Closing this issue as obsolete. If the problem is still relevant to you, feel free to comment or open another issue.

quaintdev, 2021-02-04:
I am facing the exact same issue. The upload is stuck forever.
@SimahoJr  could you possibly share the script as a gist?

quaintdev, 2021-02-04:
Turns out adding a delay of 3 seconds like below in microupload.py solves the issue
```
            time.sleep(3) 
            files.put(remote_path, fd.read())
```

frederic-bonjour, 2021-07-25:
Hi,
Adding `time.sleep(1)` in script _microupload.py_, as mentioned above, worked for me!
Thanks a lot!

vlasovskikh, 2021-08-05:
There is #142 that proposes one way to fix it. I've left my review comments there suggesting yet another approach.

AnAvgGuy, 2022-01-08:
> There is #142 that proposes one way to fix it. I've left my review comments there suggesting yet another approach.

I have the same issue now, Ive tried thay but the line I get pointed out debugging is "raise DocopExit()", any idea what my problem might be?

marcosdantas1897, 2023-11-02:
Estou enfrentado este mesmo erro ao tentar usar o NodeMCU ESP8266 com Micropython no Pycharm, consigo ate fazer upload do código uma vez, mas se for tentar novamente a execução simplesmente não finaliza me deixando apenas com isso na tela:
Connecting to COM3
Uploading files: 0% (0/1)
C:\Users\Marcos\PycharmProjects\pythonProject2\main.py -> main.py

Se alguém tiver alguma sugestão agradeço.
Segue a imagem da minha tela de codificação,
[image]

SimahoJr, 2023-11-05:
> Estou enfrentado este mesmo erro ao tentar usar o NodeMCU ESP8266 com Micropython no Pycharm, consigo ate fazer upload do código uma vez, mas se for tentar novamente a execução simplesmente não [...]

Try:
1. Press the boot button while uploading 
2. Make sure you have the right firmware (micropython)