JetBrains/intellij-micropython #83: add follow serial output to flash script

thinkberg, 2019-02-15:
I added this little snippet at the end of the main() function in microupload.py to aid debugging:

```python
while True:
    line = board.serial.readline()
    sys.stdout.buffer.write(line)
    sys.stdout.buffer.flush()
```

In intelliJ this plugin is also ignoring the "excluded files", so it tries to flash 48 files where only three are sources. It would be helpful if the plugin would use the sources root as the base for microupload.py instead of the project root.

vlasovskikh, 2020-06-05:
@thinkberg Sorry it took me so long to get back to this project.

I didn't quite get the idea behind your suggested code snippet. microupload.py already has an upload progress indicator implemented in `progress()`. Could you elaborate a bit?

I like your second idea about using the source root as the base directory if you try to upload files or directories located inside it. Follow #81 for updates.

thinkberg, 2020-06-05:
No worries. I don't remember why I wanted to do the first part. I think the reason is not to have a progress bar, but to immediately see output from the device as soon as it has been flashed. Otherwise you lose some important info.

vlasovskikh, 2020-06-05:
@thinkberg I see your point now. I've just tried it with your snippet and it does indeed help you with seeing any your program produced for the terminal. At the same it's still not a substation for a terminal window with REPL. Ideally, it should react on Control+C by bringing you into a REPL session where Control+D soft-resets your device. Having it only to print stdout seems a bit limited and could confuse users who expect REPL.

There might be a way to run "Flash /path/to/your/file.py" run configurations in the terminal window and continue by offering you full REPL provided via microrepl.py. We have something like that for regular Python run configurations in PyCharm. There is an option when you edit a Python run configuration: "Emulate terminal in the output console".

How do you feel about it? Should I re-word your issue into an enhancement request like that?

thinkberg, 2020-06-06:
Please do. I am totally with you regarding the issues with that solution. Feel free to take what is useful and change what may be solved better.

mytechnotalent, 2022-04-15:
There is no way to output any `print(''hello')` capabilities when you use the run main.py.  Any workaround?

closed 2025-08-16 by thinkberg