JetBrains/intellij-micropython #84: Flash in Pycharm does not work, but Ampy does

omiq, 2019-02-15:
Get a ```could not enter raw repl``` message, whereas the same files and connection works in Ampy

Using Mac and latest Pycharm and latest MicroPython plugin

In Ampy I use:

```ampy -d 0.5 -p /dev/tty.SLAB_USBtoUART put main.py```

dmopalmer, 2019-02-18:
If flashing can't enter raw repl mode, it usually means that you have a terminal open to the REPL.

Close the terminal and try again.

omiq, 2019-02-18:
Yeah tried that, as I say other approaches work, repl part works in the plugin/pycharm, just not the flashing part fails with this message. It’s not because the device is in use.

dmopalmer, 2019-02-18:
To be more explicit, I always forget to close the REPL in the terminal tab in the PyCharm IDE window, and get that error.  (If you go to the REPL you will see that it was put into raw mode.)

It would be nice if the plug-in could temporarily suspend the terminal connection while it uploads etc.  But that is a feature request for the future, and is probably not easy.

omiq, 2019-02-18:
I get this error regardless, there’s definitely nothing connected at the time. 

I’m wondering if it needs a longer delay or something

tlvlp, 2019-02-20:
This problem might be related to this issue https://github.com/pycampers/ampy/issues/19
You are using ampy with the -d 0.5 flag that solves the problem by adding a delay 

From the [ampy readme](https://github.com/pycampers/ampy):
> Fix for macOS users' "Could not enter raw repl"; try 2.0 and lower from there:
AMPY_DELAY=0.5

Found the same delay for this plugin at [intellij-micropython/scripts/microupload.py](https://github.com/vlasovskikh/intellij-micropython/blob/master/scripts/microupload.py)

```
def wait_for_board() -> None:
    """Wait for some ESP8266 devices to become ready for REPL commands."""
    time.sleep(0.5)
```
Maybe the 0.5s sleep timer is ok when using ampy but not enough when using the plugin?

omiq, 2019-02-20:
Makes sense, will try!

tlvlp, 2019-02-20:
I just started using this plugin and occasionally run into the same issue. 

For me re-uploading the MicroPython firmware on the board works for a random number of flash operations until I get the same error again. It's very random and cumbersome even when automated with a bash script as it takes minutes to update.

So were you able to modify the plugin somehow?

tlvlp, 2019-02-21:
I just noticed that the issue only occurs when I try to flash boot.py via the plugin. Once I do that no upload works via this plugin until I refresh the firmware.

omiq, 2019-02-25:
> I just noticed that the issue only occurs when I try to flash boot.py via the plugin.

This is the answer! In my tests, boot.py has been the first file. Excluding boot.py does the trick 💯

closed 2019-02-25 by omiq

tlvlp, 2019-02-25:
I also noticed that in many cases this plugin (or more like ampy in general) refuse to upload certain files even after a firmware update. 
These can be worked around by including them in a folder and uploading the whole folder. 
I also just learned that ampy is no longer supported by adafruit so there will likely be no solution for these problems on this side.