JetBrains/intellij-micropython #85: Accelerometer not working on Micro:bit

mrfabroa, 2019-02-17:
Accelerometer on the Micro:bit does not work when flashing with the MicroPython plugin in Pycharm.   Readings from the accelerometer remain at 0 for all axis as indicated by printing to the REPL and displaying to the device.   uFlash 1.08 was used in this case.  

This issue does not occur when flashing the same code to the device using the MU Editor, where accelerometer output works as expected.   The same Python 3.6 interpreter was used when testing both (Pycharm, MU Editor) environments.

```
from microbit import *

while True:
    sleep(1000)
    x_val = accelerometer.get_x()
    print(str(x_val))
    display.scroll(str(x_val))
```

carlosperate, 2019-03-03:
It's possible this is because this plugin is using an old version of uFlash, as Mu uses a newer version.
PR https://github.com/vlasovskikh/intellij-micropython/pull/79 should fix this issue.

fzymek, 2019-03-30:
:+1: oh wow, this would be awesome to have it

vlasovskikh, 2020-06-05:
@mrfabroa I've just accepted PR #79 by @carlosperate. I'll release a new version of the plugin soon.

closed 2020-06-05 by vlasovskikh