JetBrains/intellij-micropython #87: No context aware documentation for micropython

Walgensrl, 2019-03-13:
Hi everybody,
I'm using Pycharm 2018.3.5 CE with Micropython plugin in Windows 7.
Everything works fine: I can access REPL for my NUCLEO board, I can flash file to my board.
The problem is I can't see Mycropython documentation for my code, just standard Python. If I place the cursor on my code snippet and press CTRL+Q it says no documentation found.

What am I doing wrong?
Many thanks.

Walter

[image]

vlasovskikh, 2020-06-05:
@Walgensrl Sorry for my late reply. It's a bug in the intellij-micropython plugin. We still don't have type hinted stubs with documentation for Pyboard-specific MicroPython modules, including `pyb`.

vlasovskikh, 2020-06-05:
This issue is a duplicate of #16.

closed 2020-06-05 by vlasovskikh