JetBrains/intellij-micropython #89: REPL Console broken in PyCharm 2019.1(.1)

markprins, 2019-04-06:
The 2019.1(.1) release of PyCharm seems to have altered the way the terminal is used. It has more broad problems (like ctrl-c not working anymore, which was fixed in 2019.1.1). However it also seems to be handling output differently and triggers warnings.

As described in issue #50 when a unit is hard-resetting (or in my case returning from deep sleep) non-ascii characters are displayed. Probably because the serial connection tries to display stuff it shouldn't. 

However in pycharm 2019 it triggers an exception which essentially halts the REPL console.

Previous console output:
[image]

In this case the console will keep working (not shown in screenshot) and continues execution of code (and displaying output)

New console output:
[image]

In this case the output completely stops. Which is the biggest annoyance. Reverting to any 2018 version of PyCharm resolves the problem

Further it seems to _sometimes_ halt execution of the code as well. However I have not reliably reproduced this completely and it could be something due to my code.

Used hardware:
- Wemos Mini D1 v3.1.0 - ESP8266 
Used software:
- Pycharm v2019.1.1
- OSX High Sierra
- MicroPython v1.0.11

hlovatt, 2019-05-06:
Doesn't work for me with: PyCharm 2019.1.1 CE, Mojava 10.14.4, and pyboard.

If `from machine import Pin; led = Pin(2, Pin.OUT)` I get:

```
Traceback (most recent call last):
  File "main.py", line 18, in <module>
  File "main.py", line 6, in main
TypeError: can't convert 'int' object to str implicitly
MicroPython v1.10 on 2019-01-25; PYBv1.1 with STM32F405RG
```

Where line 6 referenced in error is `led = ...` line.

OptimusGREEN, 2020-05-04:
I just cant get this to work on high sierra at all

vlasovskikh, 2020-06-05:
@markprins Sorry for my very late reply. Is it still an issue for you? I cannot reproduce it now with my ESP8266 running MicroPython 1.9.3 and PyCharm 2020.1.

vlasovskikh, 2020-06-05:
@hlovatt It looks like a different problem, not related to the client-side exception mentioned by the original reporter. Please file another issue. I don't have a Pyboard device at the moment so I'm unable to reproduce it.

vlasovskikh, 2020-06-05:
@OptimusGREEN Please file another issue and give more details about your setup and the steps to reproduce your problem.

vlasovskikh, 2020-06-13:
Closing this issue as obsolete. Please feel free to comment it if the problem persists.

closed 2020-06-13 by vlasovskikh

hlovatt, 2020-06-14:
Sorry for my slow response. I tried out the plug-in and found that it worked to download to a PyBoard and that I could run the REPL from the plugin on the PyBoard. However the import of pyb failed (can’t find pyb). This does make development annoying since virtually all the lines show in PyCharm as an error and therefore you only find, real, problems when you Run. Is it expected that pyb be available to PyCharm?

vlasovskikh, 2020-06-14:
@hlovatt Missing code insight for `pyb` is a known problem, please follow #16. I look for volunteers to contribute type hints for `pyb`.

hlovatt, 2020-06-25:
Having used plugin more there is a problem still. You have to reset the PyBoard before you can download new code. After you have reset you then have to restart the MicroPython terminal. 

If you don't reset the error is the run window gets to:
```
Connecting to /dev/cu.usbmodem20803374594B2
Uploading files: 0% (0/1)
/Users/lov080/Google Drive/Python/examples/TestPyBoardTypeshed/main.py -> main.py
```
and then hangs. If the then reset PyBoard the run window says:
```
Traceback (most recent call last):
  File "/Users/lov080/Google Drive/Python/venv/TestPyBoardTypeshed/lib/python3.7/site-packages/serial/serialposix.py", line 493, in read
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/lov080/Library/Application Support/JetBrains/PyCharmCE2020.1/plugins/intellij-micropython/scripts/microupload.py", line 139, in <module>
    main(sys.argv[1:])
  File "/Users/lov080/Library/Application Support/JetBrains/PyCharmCE2020.1/plugins/intellij-micropython/scripts/microupload.py", line 79, in main
    files.put(remote_path, fd.read())
  File "/Users/lov080/Google Drive/Python/venv/TestPyBoardTypeshed/lib/python3.7/site-packages/ampy/files.py", line 208, in put
    self._pyboard.enter_raw_repl()
  File "/Users/lov080/Google Drive/Python/venv/TestPyBoardTypeshed/lib/python3.7/site-packages/ampy/pyboard.py", line 189, in enter_raw_repl
    data = self.read_until(1, b'raw REPL; CTRL-B to exit\r\n>')
  File "/Users/lov080/Google Drive/Python/venv/TestPyBoardTypeshed/lib/python3.7/site-packages/ampy/pyboard.py", line 163, in read_until
    new_data = self.serial.read(1)
  File "/Users/lov080/Google Drive/Python/venv/TestPyBoardTypeshed/lib/python3.7/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured
```
You can then download, but also need to restart the MicroPython terminal, because the program has run without the terminal (that went away because of reset). You therefore then need to ^C to stop program and ^D to rerun, so that you can see any start up messages and/or errors.

Any chance of taking a look! 

Plugin version is 1.1.1, PyCharm is 2020.1.2 CE, MicroPython v1.12-555-g026fda605, MacOS 10.15.5.

vlasovskikh, 2020-06-26:
@hlovatt For closing the REPL session on flashing files to the device please follow #53. If several consecutive uploads *with no REPL involved* fail for you, please file a separate issue.