JetBrains/intellij-micropython #90: fail to recognize serial port

hamortar, 2019-05-07:
I want to upload a Micropython file to an attached ESP8266, but Pycharm does not recognize the serial port in my MacOS.
I am pretty sure both the port and the driver are ok, since I can use uPyCraft with no incident.
At the Preferences/Language & Frameworks/Micropython screen I have specified the device type and the serial port path, but still Pycharm does not connect

ivkond, 2019-08-23:
Please, provide more detailed steps:
- when PyCharm does not recognize the port?
- does plugin Serial Monitor able to connect to device?
- do you change connection baud rate?

vlasovskikh, 2020-06-05:
@hamortar Closing this issue as obsolete. Feel free to open another one if the problem is still relevant to you.

closed 2020-06-05 by vlasovskikh